openbmc/openbmc #2656: OP910 : Mapper Error-  Error starting event loop: Connection timed out

cdeadmin, 2017-11-20:


tomjoseph83, 2017-11-20:
This issue is opened to increase the priority of the mapper process to -5. The mapper is the dbus service discovery manager which is used pervasively by openbmc applications. During stress tests it was observed that mapper is loaded and fails leading to a chain of other failures. The plan is to set a higher priority for the mapper process.The observation with the changes was that mapper responded faster leading to an overall better BMC performance.

rfrandse, 2017-11-20:
<https://gerrit.openbmc-project.xyz/7997> Set the nice value of mapper process to -5
Resolves  #2656 Mapper Error-  Error starting event loop: Connection timed out

closed 2018-02-08 by geissonator