openbmc/openbmc #2657: Unable to power off when critical power off services fail

geissonator, 2017-11-20:
I was looking at a system that hit #2656 in the power off path.  I see a 2 issues I'd like to investigate here:

- Is there something we can do to force a power off if the power off target fails (i.e. cut power via a gpio?) - In emergency power off situations, we can't just fail and leave power on
- Why once we hit this fail do targets like "obmc-chassis-hard-poweroff.target" not work?  I had to request a power on, then power off to get it to power off correctly.

```
Nov 19 06:20:13 witherspoon ipmid[1295]: Soft off is running, so let shutdown target stop the host
Nov 19 06:20:13 witherspoon systemd[1]: Started Soft power off of the host.
Nov 19 06:20:13 witherspoon systemd[1]: Reached target Host0 (Stopping).
Nov 19 06:20:13 witherspoon systemd[1]: Reached target Host0 (Stopped).
Nov 19 06:20:13 witherspoon systemd[1]: Reached target Power0 Off (Pre).
Nov 19 06:20:13 witherspoon systemd[1]: Started Stop Power0.
Nov 19 06:20:13 witherspoon systemd[1]: Starting Wait for Power0 to turn off...
Nov 19 06:20:29 witherspoon mapper[25649]: Error finding '/xyz/openbmc_project/led/groups/power_on' service: Connection timed out
Nov 19 06:20:29 witherspoon systemd[1]: obmc-led-group-stop@power_on.service: Main process exited, code=exited, status=1/FAILURE
Nov 19 06:20:29 witherspoon systemd[1]: Failed to start Assert power_on LED.
Nov 19 06:20:29 witherspoon systemd[1]: obmc-led-group-stop@power_on.service: Unit entered failed state.
Nov 19 06:20:29 witherspoon systemd[1]: obmc-led-group-stop@power_on.service: Failed with result 'exit-code'.
Nov 19 06:20:30 witherspoon phosphor-log-manager[980]: Failed to find metadata
Nov 19 06:20:30 witherspoon phosphor-fru-fault-monitor[1191]: /xyz/openbmc_project/logging/entry/71
Nov 19 06:20:39 witherspoon pgood_wait[25696]: Error obtaining host service: Connection timed out
Nov 19 06:20:39 witherspoon systemd[1]: op-wait-power-off@0.service: Main process exited, code=exited, status=1/FAILURE
Nov 19 06:20:39 witherspoon systemd[1]: Failed to start Wait for Power0 to turn off.
Nov 19 06:20:39 witherspoon systemd[1]: Dependency failed for Chassis0 (Power Off).
Nov 19 06:20:39 witherspoon systemd[1]: Dependency failed for Power0 Host Off.
Nov 19 06:20:39 witherspoon systemd[1]: obmc-host-shutdown@0.target: Job obmc-host-shutdown@0.target/start failed with result 'dependency'.
Nov 19 06:20:39 witherspoon systemd[1]: obmc-host-shutdown@0.target: Triggering OnFailure= dependencies.
Nov 19 06:20:39 witherspoon systemd[1]: obmc-chassis-poweroff@0.target: Job obmc-chassis-poweroff@0.target/start failed with result 'dependency'.
Nov 19 06:20:39 witherspoon systemd[1]: op-wait-power-off@0.service: Unit entered failed state.
Nov 19 06:20:39 witherspoon systemd[1]: op-wait-power-off@0.service: Failed with result 'exit-code'.
Nov 19 06:20:39 witherspoon systemd[1]: Starting Assert power_on LED...
Nov 19 06:20:39 witherspoon systemd[1]: Starting OpenPOWER OCC Active Disable...
Nov 19 06:20:39 witherspoon systemd[1]: Starting Wait for Power0 to turn off...
Nov 19 06:20:39 witherspoon mapper[25695]: Error finding '/org/openbmc/control/power0' service: Connection timed out
Nov 19 06:20:39 witherspoon systemd[1]: op-power-stop@0.service: Main process exited, code=exited, status=1/FAILURE
Nov 19 06:20:40 witherspoon systemd[1]: Started OpenPOWER OCC Active Disable.
Nov 19 06:20:46 witherspoon phosphor-ledmanager[975]: Asserting LEDs
Nov 19 06:20:46 witherspoon systemd[1]: Started Assert power_on LED.
 
 
# Some debug after the fact:
 
[witherspoon]# mapper get-service /xyz/openbmc_project/led/groups/power_on
:1.4
 
[witherspoon]# mapper get-service /org/openbmc/control/power0             
:1.90
```

Power off request when in fail state (it just sits here):
```
Nov 19 08:07:37 witherspoon phosphor-chassis-state-manager[1290]: Change to Chassis Requested Power State
Nov 19 08:07:37 witherspoon systemd[1]: Starting OpenPOWER OCC Active Disable...
Nov 19 08:07:37 witherspoon systemd[1]: xyz.openbmc_project.Ipmi.Internal.SoftPowerOff.service: Main process exited, code=killed, status=15/TERM
Nov 19 08:07:37 witherspoon systemd[1]: Stopped Soft power off of the host.
Nov 19 08:07:37 witherspoon systemd[1]: xyz.openbmc_project.Ipmi.Internal.SoftPowerOff.service: Unit entered failed state.
Nov 19 08:07:37 witherspoon systemd[1]: xyz.openbmc_project.Ipmi.Internal.SoftPowerOff.service: Failed with result 'signal'.
Nov 19 08:07:38 witherspoon systemd[1]: Started OpenPOWER OCC Active Disable.
```

amboar, 2017-11-20:
FWIW, there's now `obmcutil chassiskill` which does what you want: It just whacks all of the defined power GPIOs, de-asserting them. This immediately powers the host of, and triggers the PGOOD monitoring infrastructure which creates an error log and recovers the BMC state.

Obviously `chassiskill` is a work-around for extreme circumstances. What you're essentially considering is how to correctly do exception handling in and across systemd units/targets. I feel like it will have non-obvious complexities. I wonder if we could architect differently to minimise this problem.

geissonator, 2017-12-14:
Long term goal here is a single process that is started by a service file which cuts power to the chassis (i.e. no dbus/mapper dependencies).  That will be done as a part of the overall refactoring of the power control application.  We could temporarily do a .bbappend that calls obmcutil chassiskill on error for this target but I think for now we've addressed the main performance issues in the code base.

geissonator, 2018-07-12:
We'll track the solution for this one in https://github.com/openbmc/openbmc/issues/769.

closed 2018-07-12 by geissonator