openbmc/openbmc #2660: MFG:Witherspoon, 1742Ec, Inventory.Error.NonfunctionalError against 3 fans, amber LEDs ON

cdeadmin, 2017-11-20:


msbarth, 2017-11-20:
Increasing the allowed out-of-range time to 30sec for large fan speed changes where the fan hardware was unable to reach the allowed minimum target speed within 20sec.

rfrandse, 2017-11-20:
<https://gerrit.openbmc-project.xyz/8007> wspoon: Allow fans 30sec before nonfunctional
Resolves  #2660 Inventory.Error.Nonfunctional : Error against 3 fans, amber LEDs ON

amboar, 2017-11-21:
> Increasing the allowed out-of-range time to 30sec for large fan speed changes where the fan hardware was unable to reach the allowed minimum target speed within 20sec.

We can configure the fans how we want them to perform if that's of interest. There are knobs for controlling the rate of change.

msbarth, 2017-11-22:
What are you referring to as "knobs"?

This fan hardware coupled with the max31785 has struggled in the past to meet the time allowed for larger deltas of speed changes. I've investigated in the past different combinations of the FREQ and RAMP bits within the MFR_FAN_CONFIG and was unsuccessful at tweaking the max31785 to ramp these fans quicker.

Is there something else you're suggesting as an option? I'd be interested in running some poc tests.

amboar, 2017-11-26:
>  I've investigated in the past different combinations of the FREQ and RAMP bits within the MFR_FAN_CONFIG and was unsuccessful at tweaking the max31785 to ramp these fans quicker.

These were the knobs I was referring to. If they don't do what we need then I don't have any other suggestions.

closed 2017-11-28 by geissonator