openbmc/openbmc #2661: setting date on bmc command line results in incorrect result

geissonator, 2017-11-20:
This is with latest v2.  Am I doing something wrong here?

```
[witherspoon]# date +%Y%m%d -s "20171120"
20180817

[witherspoon]# date
Fri Aug 17 11:20:09 UTC 2018
```

mine260309, 2017-11-21:
@geissonator The busybox `date` is NOT GNU `date`, and the accepted formats are quoted below:

> //usage:     "\nRecognized TIME formats:"
> //usage:     "\n	hh:mm[:ss]"
> //usage:     "\n	[YYYY.]MM.DD-hh:mm[:ss]"
> //usage:     "\n	YYYY-MM-DD hh:mm[:ss]"
> //usage:     "\n	[[[[[YY]YY]MM]DD]hh]mm[.ss]"
> //usage:	IF_FEATURE_DATE_COMPAT(
> //usage:     "\n	'date TIME' form accepts MMDDhhmm[[YY]YY][.ss] instead"
> //usage:	)

We have `FEATURE_DATE_COMPAT` defined, so the accepted formats becomes `MMDDhhmm[[YY]YY][.ss]` in your case, and it's interpreted as:
```
MM=20
DD=17
hh=11
mm=20
```
So it becomes `2017-20-17:11:20`, where month 20 becomes 1year+8m, so it eventually becomes `2018-08-17:11:20`

If you want to set date to 2017-11-20, you can `date "2017-11-20"`

bjwyman, 2017-11-29:
Why does that not agree with the help `date` provides?
```
# date -h
date: invalid option -- 'h'
BusyBox v1.24.1 (2017-11-16 16:51:25 CST) multi-call binary.

Usage: date [OPTIONS] [+FMT] [TIME]
```

mine260309, 2017-11-29:
As quoted above, busybox data does not support the +%Y%m%d format.

bjwyman, 2017-11-29:
It becomes more obvious if one reads the BusyBox Command Help:
https://busybox.net/downloads/BusyBox.html

gtmills, 2018-10-12:
@geissonator @mine260309 Can we close?

mine260309, 2019-04-08:
This should be closed.

closed 2019-10-08 by gtmills