openbmc/openbmc #2665: Witherspoon Power AI needs PowerSuppyRedundancy to default to true

cdeadmin, 2017-11-21:


geissonator, 2017-11-21:
The power management subsystem on openpower platforms uses this IPMI sensor, PowerSuppyRedundancy, to determine it's policy when it detects a power supply has failed.  The default for this currently is set to "Disabled".  This means if a power supply fails, the OCC will not take corrective actions like throttling the GPU's or processors to maintain n level power settings.

It has been determined that the better default here is to have this "Enabled" so that the OCC does do these things by default when a power supply fails.

rfrandse, 2017-11-21:
<https://gerrit.openbmc-project.xyz/8030> Change default of PowerSupplyRedundancy to enabled
Resolves  #2665 Default PowerSuppyRedundancy to be Enabled

closed 2017-11-29 by geissonator

rfrandse, 2018-01-18:
<https://gerrit.openbmc-project.xyz/8043> Add sensor entry for power supply redundancy enabled.
Resolves  #2665 Default PowerSuppyRedundancy to be Enabled