openbmc/openbmc #2666: Software updaters: re-evaluate RedundancyPriority backup methods

mtritz, 2017-11-21:
Currently, the BMC and host software updaters store RedundancyPriority values to Cereal files so that they can be restored after a reboot. 

1.) Should these priority values be stored to environment variables?

2.) If so, why bother storing the values to file in the first place?

3.) Also, what's the best way to store and restore values from environment variables, accounting for the fact that we keep two sets of them - u-boot-env and alt-u-boot-env? Note that this makes reading the values from file potentially problematic:
```[witherspoon]# fw_setenv mytest true
[witherspoon]# cat /dev/mtd/u-boot-env | grep mytest
mytest=true
[witherspoon]# fw_setenv mytest false
[witherspoon]# cat /dev/mtd/u-boot-env | grep mytest
mytest=false
mytest=true

mtritz, 2017-12-05:
The real question that this issue is trying to answer is "What should be happening on a host factory reset?"

Suppose we have a machine with two host versions uploaded - one active, one not. Currently, the host factory reset will clear the persistence files with which we store the priority information for these versions. The next time the host software updater is started, both versions are left with priority of 255, and one of the two (maybe it's random, maybe it's based on upload order) is selected as the active host version.

Is this the desired behavior? Would it be better to instead retain the version priority through a reset, or is clearing that setting under the scope of "restoring factory settings"?

rfrandse, 2017-12-16:
<https://gerrit.openbmc-project.xyz/8017> Host updater: Back up version priority in environment variables
Resolves  #2666 Software updaters: re-evaluate RedundancyPriority backup methods

closed 2018-01-23 by geissonator