openbmc/openbmc #2669: PowerSupplyRedundancy must have integer type instead of string

cdeadmin, 2017-11-23:


rfrandse, 2017-11-23:
@dkodihal comments:
The reason that you're always seeing this Enabled is not because the interface is defined as a string (although I'm not sure either why this defined as a string in the old Barreleye code). The ipmi code has translation logic to convert these strings ("Enabled"/"Disabled") to appropriate offsets being set/unset. This translation logic was broken by this commit - https://gerrit.openbmc-project.xyz/#/c/8030/. Dhruv is working in fixing this correctly.

dhruvibm, 2017-12-14:
# **Status**
Fixed

geissonator, 2017-12-14:
The overall issue with power supply redundancy and the ability to set it was fixed in the following commits:
https://gerrit.openbmc-project.xyz/#/c/8166/
https://gerrit.openbmc-project.xyz/#/c/8106/
https://gerrit.openbmc-project.xyz/#/c/8031/

closed 2017-12-14 by geissonator