openbmc/openbmc #2670: [Witherspoon] : User initiated dump request throwing "501 Not Implemented" error after dump service restart

cdeadmin, 2017-11-24:


ngorugan, 2017-11-24:
With changes made for issue:1496 we were started using "well-known" name of the service during the method call instead of unique name.
When the service restarts we get new unique name and this is getting appended to bus_map list,instead we need update existing owned_name entry.

ngorugan, 2017-11-24:
https://gerrit.openbmc-project.xyz/#/c/8056 Update exsting owned_name entry in
bus_map with newly created unique name when serive restarts
Fixes #2670 User initiated dump request throwing "501 Not Implemented" error 
after dump service restart

rfrandse, 2017-11-24:
<https://gerrit.openbmc-project.xyz/8056> Update exsting owned_name entry in bus_map with newly created unique name when serive restarts
Fixes  #2670 [Witherspoon] : User initiated dump request throwing "501 Not Implemented" error after dump service restart

ngorugan, 2017-12-01:
Changes were pushed  using below given commit.
https://gerrit.openbmc-project.xyz/#/c/7579/

rfrandse, 2017-12-06:
<https://gerrit.openbmc-project.xyz/7579> server: Serve well-known bus names
Resolves  #2670 [System] : User initiated dump request throwing "501 Not Implemented" error after dump service restart

closed 2017-12-15 by geissonator