openbmc/openbmc #2671: Phosphor-networkd:- Interface specific Default Gateway.

ratagupt, 2017-11-24:
We need to provide the provision for interface specific default gateway instead of system specific.

AlexanderAmelkin, 2018-03-21:
Are you planning to implement policy-based routing in openbmc? Otherwise this is useless. Linux (unless policy based routing is set up) have no support for multiple gateways. Only a single, default, gateway is supported and used by linux routing engine.

Can you provide use cases for per-interface gateways please?

Please note that IPMI specification doesn't declare any means for setting up routing in general and routing policies specifically.

AlexanderAmelkin, 2018-03-21:
I'd say that unless we want policy-based routing, understand the use cases for it, and realize the means we're going to have for configuring it and integrating that configuration with IPMI, we don't need those `Gateway` properties within `/xyz/openbmc_project/network/<interface>/ipv4` objects, and that those must be eradicated to prevent misunderstanding.

Ramkosh, 2018-03-22:
While gateway assignment via IPMI is a separate issue (although a relative one), the current implementation and handling of the Gateway property of `xyz.openbmc_project.Network.IP` is somewhat confusing and possibly incorrect one.
Right now we either have this property set to zeroed value (`0.0.0.0`) or an empty string.

As Alexander said, we either need to determine the proper use case or remove the property entirely.

AlexanderAmelkin, 2019-04-01:
As far as I can see, we still have Gateway within xyz.openbmc_project.Network.IP interface.
If I recall right, there is some activity in WebUI project in regard to this. Anyway, I think this must remain open.

ratagupt, 2019-04-27:
Alexander, We are removing the gateway from the IP interface.

closed 2019-11-26 by stale[bot]