openbmc/openbmc #2673: [xCAT] Did not get new dump ID from response of creating dump

xuweibj, 2017-11-27:
Version:
```
mid05tor12cn11: 82087175 Host    Active(*)  IBM-witherspoon-ibm-OP9_v1.19_1.83
mid05tor12cn11: 608e9ebe BMC     Active(*)  ibm-v2.0-0-r9-0-gecda565
```

For the version "ibm-v1.99.10-113-g65edf7d-r10-0-gcdf7635 &  IBM-witherspoon-ibm-OP9_v1.19_1.73", when we create a new dump, will return the ID in the "data". But we could not get on the current version.

```
# curl -k -b cjar -X POST -H "Content-Type: application/json" -d '{"data":[]}' https://172.11.139.11/xyz/openbmc_project/dump/action/CreateDump
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
```

Why change this? And will not return the ID in the future?

@whowutwut @zet809

whowutwut, 2017-11-27:
@xuweibj This is a known issue: https://w3.rchland.ibm.com/projects/bestquest/?defect=SW407893 

It has been fixed in 1742Ee driver, `ibm-v2.0-0-r13-0-g3a160ac` 

I have verified today: 
```
[root@briggs01 910.1746.20171124a_1742Ee]# rinv $NODE firm | grep -i BMC
mid05tor12cn16: BMC Firmware Product: ibm-v2.0-0-r13-0-g3a160ac (Active)*
mid05tor12cn18: BMC Firmware Product: ibm-v2.0-0-r9-0-gecda565 (Active)*
[root@briggs01 910.1746.20171124a_1742Ee]# rspconfig $NODE dump -g
mid05tor12cn16: [30] success
mid05tor12cn18: [] success
```

mdmillerii, 2017-11-27:
That should have said fixed in User-initiated dump REST request does not respond with dump ID #2592 .

closed 2017-11-28 by xuweibj