openbmc/openbmc #2675: P8 ipmi: unable to receive eSEL

mine260309, 2017-11-28:
On P8 systems, hostboot send eSEL with `NETFUN_OEM` (`0x32`) .
In OpenBMC, https://github.com/openbmc/openpower-host-ipmi-oem/commit/64994eab31d026aa9155b45d8f84a2950e1c9fb4 changes it to `NETFUN_IBM_OEM` (`0x3a'), which breaks the function.

The netfun should be changed to a configuration parameter so that it can be configured correctly for P8/P9 systems.

tomjoseph83, 2017-11-28:
@mine260309 The hostboot has changed the NetFn from 0x32 to 0x3A. I don't see a need for OpenBMC to support two netfuns for the same command. Can the change be done on HB side for P8?

@bradbishop Thoughts?

chiabinglee, 2017-11-29:
There are two branches in hostboot code,
In `master` branch:
```
    NETFUN_AMI = (0x32 << 2),
    NETFUN_IBM = (0x3a << 2),
...
   inline const command_t partial_add_esel(void)
    { return std::make_pair(NETFUN_AMI, 0xf0); }
```

In `master-p8` branch:
```
    NETFUN_AMI = (0x32 << 2),
    NETFUN_IBM = (0x3a << 2),
...
    inline const command_t partial_add_esel(void)
    { return std::make_pair(NETFUN_AMI, 0xf0); }
```

As download master-p8 branch for P8 system, the esel command still using 0x32 NetFn.

closed 2019-03-28 by mine260309