openbmc/openbmc #2676: Clear input fault error when input fault fixed

bjwyman, 2017-11-28:
The power supply monitor (currently Witherspoon specific) should mark any input fault errors as Resolved = true if/when an input fault is fixed.

If a power supply is unplugged, a fault bit will turn on, resulting in an input fault error getting logged. When this occurs, the enclosure fault LED(s) will turn on. If the power supply is then plugged in to resolve that error, the error that was logged should get marked as resolved. If the power supply input fault was the only error driving the enclosure fault LED(s), then the enclosure fault should get deactivated.

bjwyman, 2017-12-07:
- [ ] Create a function to call for resolving an error given call out and message
- [ ] Create a function to be used for setting the Resolved property on error entry
- [ ] Call the function to resolve the input fault error when the fault bits turn off.

rfrandse, 2017-12-07:
<https://gerrit.openbmc-project.xyz/8165> Fill in the function for resolving errors
Resolves:  #2676 Clear input fault error when input fault fixed

closed 2017-12-12 by geissonator