openbmc/openbmc #2677: MPV:OP910:ECC error (corruption ) on HBEL partition after code update

cdeadmin, 2017-11-28:


anoo1, 2017-11-28:
The host fails to boot with error:

```
--== Welcome to Hostboot hostboot-e223708/hbicore.bin ==--

  4.01944|secure|SecureROM valid - enabling functionality
  4.01947|secure|Booting in non-secure mode.
  9.50263|ISTEP  6. 5 - host_init_fsi
 10.07871|ISTEP  6. 6 - host_set_ipl_parms
 10.18656|ECC error in PNOR flash in section offset 0x00026000

 10.18660|System shutting down with error status 0x60F
```

The offset 0x26000 corresponds to the HBEL partition:
https://github.com/open-power/pnor/blob/21d66daff697f8c13fb067340ca621e9208ab1ea/p9Layouts/defaultPnorLayout_64.xml

But even after code update, the system fails to boot.

anoo1, 2017-11-28:
After enabling traces in mbox, the partition that has the ECC error is the GARD partition. The confusion with the offsets is that the mbox code assumes a continuous pnor layout, but there's a gap in between the first partition (part) and the second one (HBEL):

Pnor layout as defined in the xml:
```
TOC@0x00000000 Partitions:
-----------
ID=00            part 00001000..00002000 (actual=00001000)
ID=01            HBEL 00008000..0002c000 (actual=00024000) [ECC]
ID=02           GUARD 0002c000..00031000 (actual=00005000) [ECC]
```

Pnor layout modeled by mbox:
```
TOC@0x00000000 Partitions:
-----------
ID=00            part 00001000..00002000 (actual=00001000)
ID=01            HBEL 00002000..00026000 (actual=00024000) [ECC]
ID=02           GUARD 00026000..0002b000 (actual=00005000) [ECC]
```

Will update mbox so that the offsets reflect the actual number from the xml.

sjitindarsingh, 2017-12-01:
I don't think this is a problem with the flash layout. Just looks like the GAURD partition hasn't been ECCed correctly to me. From memory a code update doesn't fix ECC errors in the GUARD partition, but I could be wrong.

anoo1, 2017-12-01:
Correct, the commit https://gerrit.openbmc-project.xyz/#/c/8089/ is just cosmetic.
Unfortunately we haven't been able to find a way to reproduce the ECC corruption.
@sjitindarsingh , Dan wanted to understand in more detail the write flow from the bmc mboxd perspective, for example what would happen if the host powered off in the middle of a write, does mboxd cache the value, discards the data, or writes to flash what has sent so far?
Per Dan, the corruption appears to be towards the end of a page.

sjitindarsingh, 2017-12-04:
To reproduce the ECC corruption you can just edit the GUARD file so that it's not ECC cleared. From my testing updating the pnor DOESN'T resolve GUARD ecc errors.

I would point Dan towards https://github.com/openbmc/mboxbridge/blob/master/Documentation/mbox_protocol.md which pretty well summarises the interaction between the host and bmc re writes.
Basically nothing is guaranteed to be written back until the host gets a success response from the flush command. Generally the host call flush at the end of a write. So basically if the host goes away anywhere during a write it can be expected nothing was written back to flash. However it will probably persist in the bmc ram (in the window caches), but isn't guaranteed to.

anoo1, 2018-01-04:
A gard clear enhancement was put in place with issue: https://github.com/open-power/boston-openpower/issues/560.
Plus the change to display the offsets: https://gerrit.openbmc-project.xyz/#/c/8089/
Closing defect.

closed 2018-01-04 by anoo1