openbmc/openbmc #2682: Enhance CI test bucket (QEMU and HW)

geissonator, 2017-11-28:


geissonator, 2018-01-03:
Current CI bucket is https://github.com/openbmc/openbmc-test-automation/blob/master/test_lists/HW_CI

Basically a REST api test, a power on, and a check for any application failures.

I think we should add:
- Verify no error logs generated
- Verify SSH and IPMI interfaces
- Verify the load averages in "uptime" do not exceed some upper limit (1 maybe?) - performance testing
- Verify code update from the image under test still works (ideally find a way to not require a full code update)
- Verify OCC active once host booted
- Add some additional REST api tests to get as much coverage as possible
- Verify a BMC reset with the host up (ensure proper states, no errors, and occ active)

gkeishin, 2018-01-09:
@mtritz let me know if u need more info. We got most of them as a test case , so all you may need to do is create a test_list/QEMU_CI and include those TC's .

anoo1, 2018-03-14:
Pending commits:
https://gerrit.openbmc-project.xyz/#/q/status:open+project:openbmc/openbmc-test-automation+branch:master+topic:2682

gkeishin, 2018-05-28:
@ssombar We can abandon the gerrit and start fresh since a bunch of code restructure and base is done. So

gkeishin, 2018-05-28:
- Add a whitelist error log such as power supply or any on demand. the best bet via an argument passed by the user explicitly
- Check for error and fail if there is a new error but ignore if the error generated is in the whitelist

rfrandse, 2018-08-16:
moved to https://github.com/ibm-openbmc/dev/issues/74

closed 2018-08-16 by rfrandse