openbmc/openbmc #2684: Dreport FFDC ehancement needed

gkeishin, 2017-11-29:
```
bash-4.1$ ls obmcdump_1_1511947103
bmc-state.log      cpuinfo         dreport.log          hostnamectl.log  inventory.log  iplink.log          meminfo           os-release   top.log
chassis-state.log  disk-usage.log  failed-services.log  host-state.log   ipaddr.log     journal-pretty.log  obmc-console.log  summary.log  uptime.log
bash-4.1$
```

Suggestion for fix and improvements:

- [x]  State data needs minor clean up
``` 
bash-4.1$ cat bmc-state.log
s "xyz.openbmc_project.State.BMC.BMCState.Ready"
bash-4.1$ cat host-state.log
s "xyz.openbmc_project.State.Host.HostState.Off"
bash-4.1$
bash-4.1$ cat chassis-state.log
s "xyz.openbmc_project.State.Chassis.PowerState.Off"
bash-4.1$
```
remove the `s` extra string out from there

- [x]  Add the `ip route show` as well in ipaddr.log

- [x] iplink.log doesnt seems to be need as ipaddr.log has that info

 - [x]  Fix this one
```
bash-4.1$ cat failed-services.log
0 loaded units listed. Pass --all to see loaded but inactive units, too.
To show all installed unit files use 'systemctl list-unit-files'.
```

- [x] journal-pretty.log  is good but for average debuggers , we need no-pager

ojayanth, 2017-11-29:
@gkeishin due to flash space limitation , need to avoid duplicate data in dreport.  For initial debug developers grep the Message part in the journal-pretty.log to understand the flow. 

 Currently  printing command output and not enabled any additional content filtering.

gkeishin, 2017-11-29:
Sure.. but the remaining other can addressed right ?

dkodihal, 2017-11-29:
@ojayanth if the flash space on Witherspoon is a concern, shouldn't you still be able to plug-in what @gkeishin has asked for other boxes, or rather, plug-out what's not needed on Witherspoon via bitbake magic?

closed 2019-04-27 by stale[bot]