openbmc/openbmc #2696: OCC make the BMC system hanging.

ryandcyu, 2017-12-01:
@eddiejames @mine260309 
I did some test about CPU, dimm temperature on zaius and BMC version is V2.0.0-75
I found that if occ1 and occ2 driver both are binds, the BMC system will be abnormal
check as top command, the system usage always keeps in 100% idle, and all process get slowly.
But if I unbind one occ (whatever occ1 or occ2), the BMC system will back to normal.
I did some test as below:

1.occ1 and occ2 driver are bind and then phosphor-hwmon-readd start to monitor occ1 and occ2. it will cause the system abnormal (100% idle) and the system gets slower.
2.If I kill one of the processes which phosphor-hwmon-readd occ. the system will back to normal.
Is it something like race condition issue in the occ driver?

CPU usage shows as below if system is normal (only one phosphor-hwmon-readd -o /gpio-fsi/cfam@0,0/sbefifo@2400/occ@1/occ-hwmon@1)
```
Mem: 159396K used, 792248K free, 14732K shrd, 10924K buff, 53404K cached
CPU:  25% usr  16% sys   0% nic  52% idle   0% io   0% irq   4% sirq
Load average: 1.60 1.62 1.31 5/147 1639
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 1612     1 root     R     6156   1%  13% phosphor-hwmon-readd -o /gpio-fsi/cfam@0,0/sbefifo@2400/occ@1/occ-hwmon@1
  716     1 root     S    13656   1%  10% /lib/systemd/systemd-journald
  987     1 messageb S     5244   1%   5% /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation
  975     1 root     R    16608   2%   5% python /usr/sbin/phosphor-mapper --path_namespaces=/org/openbmc /xyz/openbmc_project /org/open_power --interface_name
 1075     1 root     S     6152   1%   3% phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@380/nct7904@2e
 1056     1 root     S    15736   2%   3% python /usr/sbin/system_manager.py
  991     1 root     S    19580   2%   2% /usr/sbin/fan_algorithm.exe
 1073     1 root     S     6152   1%   2% phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@40/ucd90160@64
 1071     1 root     S     6152   1%   1% phosphor-hwmon-readd -o /iio-hwmon
    6     2 root     SW       0   0%   1% [ksoftirqd/0]
  997     1 root     S     1712   0%   1% inarp
  994     1 root     S    15292   2%   0% python /usr/sbin/chassis_control.py
 1104     1 root     S    32632   3%   0% power_control.exe 3000 10
 1074     1 root     S     6152   1%   0% phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@380/nct7904@2d
```
CPU usage shows as below if system is abnormal (Both of phosphor-hwmon-readd -o /gpio-fsi/cfam@0,0/sbefifo@2400/occ@1/occ-hwmon@1 and phosphor-hwmon-readd -o /gpio-fsi/cfam@0,0/hub@3400/cfam@1,0/sbefifo@2400/occ@2/occ-hwmon@2)
```
Mem: 169784K used, 781860K free, 22956K shrd, 10924K buff, 61656K cached
CPU:   0% usr   0% sys   0% nic  99% idle   0% io   0% irq   0% sirq
Load average: 1.25 1.52 1.35 6/148 1653
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 1056     1 root     S    15732   2%   0% python /usr/sbin/system_manager.py
  978     1 root     S    69324   7%   0% mboxd --flash 64M --window-size 1M
 1104     1 root     S    32632   3%   0% power_control.exe 3000 10
 1106     1 root     S    32620   3%   0% button_reset.exe
 1108     1 root     S    32620   3%   0% button_power.exe
 1086     1 root     S    32512   3%   0% flash_bios.exe
  979     1 root     S    32508   3%   0% control_bmc.exe
  716     1 root     S    30044   3%   0% /lib/systemd/systemd-journald
    1     0 root     S    26388   3%   0% {systemd} /sbin/init
  991     1 root     S    20860   2%   0% /usr/sbin/fan_algorithm.exe
  980     1 root     S    18008   2%   0% python /usr/sbin/netman.py
 1085     1 root     S    17772   2%   0% python /usr/sbin/bmc_update.py
  975     1 root     S    16608   2%   0% python /usr/sbin/phosphor-mapper --path_namespaces=/org/openbmc /xyz/openbmc_project /org/open_power --interface_name
 1007     1 root     S    15672   2%   0% python /usr/sbin/download_manager.py
 1009     1 root     S    15668   2%   0% python /usr/sbin/inventory_items.py
  985     1 root     S    15668   2%   0% python /usr/sbin/sensor_manager2.py
 1058     1 root     S    15444   2%   0% {settings_manage} /usr/bin/python -u /usr/sbin/settings_manager.py
  994     1 root     S    15292   2%   0% python /usr/sbin/chassis_control.py
  962     1 systemd- S    14936   2%   0% /lib/systemd/systemd-timesyncd
  988     1 systemd- S    13900   1%   0% /lib/systemd/systemd-networkd
 1129     1 root     S     7692   1%   0% ipmid
 1059     1 root     S     7448   1%   0% phosphor-image-updater
 1100     1 root     S     7356   1%   0% phosphor-version-software-manager
  983     1 root     S     6676   1%   0% phosphor-inventory
 1643     1 root     S     6268   1%   0% phosphor-hwmon-readd -o /gpio-fsi/cfam@0,0/hub@3400/cfam@1,0/sbefifo@2400/occ@2/occ-hwmon@2
 1062     1 root     S     6200   1%   0% phosphor-network-manager
 1008     1 root     S     6168   1%   0% phosphor-user-manager
  976     1 root     S     6160   1%   0% phosphor-settings-manager
 1612     1 root     S     6156   1%   0% phosphor-hwmon-readd -o /gpio-fsi/cfam@0,0/sbefifo@2400/occ@1/occ-hwmon@1
111

mine260309, 2017-12-04:
Reproduced on Romulus.
The easy steps to reproduce:
1. Bind both CPU's occ-hwmon;
2. Get occ-hwmon sensor's reading for every second: `watch -n 1 cat /sys/class/hwmon/hwmonX/temp1_input`

Then the issue is easily reproduced: it is very slow for watch to get sensor reading.

A guess is that occ-hwmon driver takes time to get a reading, and the `watch -n 1` queues the sensor readings requests and it is busy polling occ.

However, if only one CPU's occ-hwmon is bind, the issue won't occur.

msbarth, 2017-12-04:
FYI...With this reproduced on Romulus, this would not be related to the performance issues within fan control. @dkodihal ^

eddiejames, 2017-12-12:
Lei, what openbmc version are you running? There have been a lot of fixes to occ/sbefifo drivers in the past few months.

I tried the latest on zaius today and transfers were taking up to 10s but the bmc was still responsive. Slower than without hwmon polling, but responsive. The drivers aren't busy-waiting anywhere. In fact, the largest delays in the driver come from scheduling out to the kernel or to userspace I believe. See the trace below for a transfer taking 7 seconds, where the bulk of the time spent is just in scheduling the sbefifo timer function. That is not spent in busy time in the driver. It's not waiting for another occ/sbefifo operation either.

[  517.280000] occ[b8be5cf0] work start [217280]
[  517.280000] occ[b8be5cf0] write cmd and trigger attn [217280]
[  517.280000] sbefifo[bca51400] start write [217280]
[  517.280000] sbefifo[bca51400] finish write [217280]
[  517.280000] sbefifo[bca51400] start read [217280]
<<< 1 second between call to start the work thread and actually starting it
[  518.240000] sbefifo[bca51400] start work [218240]
[  518.240000] sbefifo[bca51400] finish work [218240]
[  518.240000] sbefifo[bca51400] done waiting for read [218240]
[  518.240000] sbefifo[bca51400] finish read [218240]
[  518.240000] sbefifo[bca51400] start write [218240]
[  518.240000] sbefifo[bca51400] finish write [218240]
[  518.240000] sbefifo[bca51400] start read [218240]
<<< same 1 second of scheduling
[  519.210000] sbefifo[bca51400] start work [219210]
[  519.210000] sbefifo[bca51400] finish work [219210]
[  519.210000] sbefifo[bca51400] done waiting for read [219210]
[  519.210000] sbefifo[bca51400] finish read [219210]
[  519.210000] occ[b8be5cf0] triggered attn, begin read first 8 bytes [219210]
[  519.210000] sbefifo[bca51400] start write [219210]
[  519.210000] sbefifo[bca51400] finish write [219210]
[  519.210000] sbefifo[bca51400] start read [219210]
<<< another second
[  520.180000] sbefifo[bca51400] start work [220180]
[  520.180000] sbefifo[bca51400] finish work [220180]
[  520.180000] sbefifo[bca51400] done waiting for read [220180]
[  520.180000] sbefifo[bca51400] finish read [220180]
[  520.180000] sbefifo[bca51400] start read [220180]
[  520.180000] sbefifo[bca51400] done waiting for read [220180]
[  520.180000] sbefifo[bca51400] finish read [220180]
[  520.180000] occ[b8be5cf0] got first 8 bytes, begin read response [220180]
[  520.180000] sbefifo[bca51400] start write [220180]
[  520.180000] sbefifo[bca51400] finish write [220180]
[  520.180000] sbefifo[bca51400] start read [220180]
[  521.160000] sbefifo[bca51400] start work [221160]
[  521.160000] sbefifo[bca51400] finish work [221160]
[  521.160000] sbefifo[bca51400] done waiting for read [221160]
[  521.160000] sbefifo[bca51400] finish read [221160]
[  521.160000] sbefifo[bca51400] start read [221160]
[  522.140000] sbefifo[bca51400] start work [222140]
[  522.140000] sbefifo[bca51400] finish work [222140]
[  522.140000] sbefifo[bca51400] done waiting for read [222140]
[  522.140000] sbefifo[bca51400] finish read [222140]
[  522.140000] sbefifo[bca51400] start read [222140]
[  523.130000] sbefifo[bca51400] start work [223130]
[  523.130000] sbefifo[bca51400] finish work [223130]
[  523.130000] sbefifo[bca51400] done waiting for read [223130]
[  523.130000] sbefifo[bca51400] finish read [223130]
[  523.130000] sbefifo[bca51400] start read [223130]
[  524.120000] sbefifo[bca51400] start work [224120]
[  524.120000] sbefifo[bca51400] finish work [224120]
[  524.120000] sbefifo[bca51400] done waiting for read [224120]
[  524.120000] sbefifo[bca51400] finish read [224120]
[  524.120000] sbefifo[bca51400] start read [224120]
[  525.120000] sbefifo[bca51400] start work [225120]
[  525.120000] sbefifo[bca51400] finish work [225120]
[  525.120000] sbefifo[bca51400] done waiting for read [225120]
[  525.120000] sbefifo[bca51400] finish read [225120]
[  525.120000] sbefifo[bca51400] start read [225120]
[  525.120000] sbefifo[bca51400] done waiting for read [225120]
[  525.120000] sbefifo[bca51400] finish read [225120]
[  525.120000] occ[b8be5cf0] got response 553 bytes [225120]
[  525.120000] occ[b8be5cf0] work finish [225120]

mine260309, 2017-12-14:
@eddiejames I just tried the latest build today and still reproduce the issue on Romulus.

>  transfers were taking up to 10s but the bmc was still responsive. Slower than without hwmon polling, but responsive.

Do you mean it could take 10 seconds to get a temp sensor reading?
From what I see on Romulus and what Ryan see on BarreleyeG2, the BMC is much slower than normal in such case.

eddiejames, 2017-12-14:
@mine260309 yes could be 10 seconds or more to read a temp sensor or other OCC sensor. Seems like it takes a long time to schedule each SBEFIFO operation. That's controlled by the kernel, not my drivers.

So far I haven't found any problem in the drivers. Nothing is holding the CPU that would slow down user code or other kernel code that I can see...

ryandcyu, 2017-12-18:
@eddiejames @mine260309 
I have modified the monitor interval from 1s to 10s, you could refer the patch https://github.com/ryandcyu/phosphor-hwmon/commit/8b95ece3e31ba28af21c8d6a4928820c6d1260bb
And the BMC response faster than before, but this is not reliable, because of CPU temp may increase rapidly at full loading, However, we just only refresh the temperature after 10 seconds. We can not cool down instantly with the fan mechanism.

eddiejames, 2017-12-18:
I'm going to make a change to sbefifo scheduling that should help a lot. On witherspoon the transfers were reduced from 1 second to 30ms. I think it's a workaround for some scheduling problem though, associated with hwmon. Not a real fix.

ryandcyu, 2018-01-22:
Hi @madscientist159
Could you also try to replicate this issue on your system?
I guess is that occ related issue as well.
According to your experience, you may know what root cause is.

anoo1, 2018-02-14:
Patches so far for this issue:
https://gerrit.openbmc-project.xyz/#/c/8351/
https://gerrit.openbmc-project.xyz/#/c/8814/

amboar, 2018-02-17:
I posted a series which is a mix of patches from @eddiejames @jk-ozlabs and myself:

http://patchwork.ozlabs.org/project/openbmc/list/?series=28746&state=%2A&archive=both

Please test it if you've hit this bug

anoo1, 2018-03-05:
Ryan reports there are no more OCC errors with patchset - https://gerrit.openbmc-project.xyz/#/c/9002/

mine260309, 2018-03-30:
With latest OpenBMC (kernel on dev-4.13), the issue is gone.
I think we can close this issue now.

closed 2018-04-18 by amboar