openbmc/openbmc #2698: OP910: Software Manager - "Error in mapper GetSubTreePath " on factory reset

cdeadmin, 2017-12-01:


anoo1, 2017-12-01:
After factory reset, the BMC updater process fails to find the chassis inventory needed to create the functional version association. This is due because the inventory was cleared with the factory reset and not re-populated by the time the BMC updater process starts. Need to add a dependency to the service. Journal snippet for reference:

```
Nov 15 01:06:49 witherspoon systemd[1]: Starting Phosphor BMC Code Update...
Nov 15 01:06:49 witherspoon systemd[1]: Started Wait for /xyz/openbmc_project/inventory.
Nov 15 01:06:52 witherspoon phosphor-gpio-presence[1210]: Updating inventory present property
Nov 15 01:06:52 witherspoon phosphor-gpio-presence[1206]: Updating inventory present property
Nov 15 01:06:52 witherspoon phosphor-image-updater[1161]: Error in mapper GetSubTreePath
Nov 15 01:06:56 witherspoon systemd[1]: Started OpenBMC Software Update Manager.
Nov 15 01:06:56 witherspoon systemd[1]: Starting Phosphor Version Software Manager...
Nov 15 01:06:57 witherspoon systemd[1]: Started Phosphor Version Software Manager.
Nov 15 01:07:00 witherspoon systemd[1]: Started Phosphor BMC Code Update.
Nov 15 01:07:00 witherspoon bmc_update.py[1201]: Running Bmc Flash Control
Nov 15 01:07:06 witherspoon systemd[1]: Started Wait for /xyz/openbmc_project/inventory/system/chassis/power_supply0.
Nov 15 01:07:06 witherspoon systemd[1]: Started Wait for /xyz/openbmc_project/inventory/system/chassis/power_supply1.
```

anoo1, 2017-12-01:
After adding service dependency:

```
Dec 01 22:20:59 witherspoon systemd[1]: Started Phosphor BMC Code Update.
Dec 01 22:21:02 witherspoon systemd[1]: Started Wait for /xyz/openbmc_project/inventory/system/chassis/powersupply0.
Dec 01 22:21:02 witherspoon systemd[1]: Started Wait for /xyz/openbmc_project/inventory/system/chassis/powersupply1.
Dec 01 22:21:02 witherspoon systemd[1]: Started Wait for /xyz/openbmc_project/inventory/system/chassis.
Dec 01 22:21:02 witherspoon systemd[1]: Starting OpenBMC Software Update Manager...
```

rfrandse, 2017-12-02:
<https://gerrit.openbmc-project.xyz/8121> Software.BMC.Updater: Wait for chassis inventory
Resolves:  #2698  Software Manager - "Error in mapper GetSubTreePath " on factory reset

closed 2018-01-18 by geissonator