openbmc/openbmc #2705: Porting for application local errors

devenrao, 2017-12-06:
Following applications need to be catered for changes done to application local errors. At present it is done for open-power-debug-collector same needs to be done to below applications

./phosphor-led-manager/elog-errors.hpp
./openpower-occ-control/elog-errors.hpp
./openpower-host-ipmi-oem/elog-errors.hpp
./phosphor-bmc-code-mgmt/elog-errors.hpp
./phosphor-host-ipmid/elog-errors.hpp
./phosphor-time-manager/elog-errors.hpp

spinler, 2017-12-06:
Can you also look at witherspoon-pfault-analysis?

rfrandse, 2017-12-19:
<https://gerrit.openbmc-project.xyz/8349> Enable obmcconsole script for elog type error
Resolves  #2705 Porting for application local errors

rfrandse, 2018-01-22:
<https://gerrit.openbmc-project.xyz/8394> op-occ-control: add native recipe to install error yaml to shared location
Partially resolves  #2705 Porting for application local errors

closed 2018-01-30 by geissonator

bjwyman, 2018-02-02:
WARNING: openpower-occ-control-1.0-r1 do_configure: QA Issue: openpower-occ-control: configure was passed unrecognised options: --disable-gen_errors [unknown-configure-option]

Seems that the --disable-gen_errors part of the recipe did not have a matching change in all the other affected repos.

devenrao, 2018-02-02:
There is a patch pending to be merged, working on it.

bjwyman, 2018-02-13:
Assuming you are referring to this up on gerrit:
https://gerrit.openbmc-project.xyz/8383