openbmc/openbmc #2709: OP910: Software Manager doesn't show fully qualified active and functional image objects

cdeadmin, 2017-12-11:


anoo1, 2018-01-04:
Software end qualified active shows only 1 when 2 are active and functional

```
$ ./rest_cmd -i wsbmc002 -o GET -u /xyz/openbmc_project/software/enumerate
Executing: GET  /xyz/openbmc_project/software/enumerate

/xyz/openbmc_project/software/enumerate
{
...
    "/xyz/openbmc_project/software/active": {
        "endpoints": [
            "/xyz/openbmc_project/software/40163480",
            "/xyz/openbmc_project/software/29c0a606"
        ]
    },
    "/xyz/openbmc_project/software/functional": {
        "endpoints": [
            "/xyz/openbmc_project/software/40163480",
            "/xyz/openbmc_project/software/29c0a606"
        ]
    }
}
```

```
$ ./rest_cmd -i wsbmc002 -o GET -u /xyz/openbmc_project/software
Executing: GET  /xyz/openbmc_project/software

/xyz/openbmc_project/software
{
    "FieldModeEnabled": 1,
    "associations": [
        [
            "functional",
            "software_version",
            "/xyz/openbmc_project/software/29c0a606"
        ],
        [
            "active",
            "software_version",
            "/xyz/openbmc_project/software/29c0a606"
        ]
    ]
}
```

joseph-reynolds, 2018-06-01:
As far as I got today, these URIs are handled by phosphor-rest-server/module/obmc/wsgi/apps/rest_dbus.py.
class ListHandler handles GET /xyz/openbmc_project/software/enumerate.
class InstanceHandler handles GET  /xyz/openbmc_project/software.

rfrandse, 2018-08-17:
moved to https://github.com/ibm-openbmc/dev/issues/103

gtmills, 2018-10-12:
@anoo1 This still a problem?

closed 2019-05-14 by stale[bot]