openbmc/openbmc #2715: Fan target sysfs files show -1 speed

msbarth, 2017-12-13:
After powering on the chassis, the fan targets were not correctly displayed within sysfs for the fan targets.
```
root@witherspoon:/sys/class/hwmon/hwmon9# obmcutil powerpgood = 1
state = 1
pgood_timeout = 10

root@witherspoon:/sys/class/hwmon/hwmon9# cat /etc/os-release ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v2.0"
VERSION_ID="ibm-v2.0-0-r19-0-ge7f2422-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v2.0"
BUILD_ID="ibm-v2.0-0-r19"

root@witherspoon:/sys/class/hwmon/hwmon9# cat /var/lib/phosphor-software-manager/pnor/ro/VERSION 
IBM-witherspoon-ibm-OP9_v1.19_1.88
	op-build-v1.19-272-g7e6959e
	buildroot-2017.08-8-g5e23247
	skiboot-v5.9.4
	hostboot-47e26a3
	linux-4.13.11-openpower1-p502d202
	petitboot-v1.6.3-p393c40d
	machine-xml-bcd4a40
	occ-51965bd
	hostboot-binaries-325a008
	capp-ucode-p9-dd2-v2
	sbe-93ed873

root@witherspoon:/sys/class/hwmon/hwmon9# tail -v fan*_target==> fan1_target <==
-1

==> fan2_target <==
-1

==> fan3_target <==
-1

==> fan4_target <==
-1
```
I was able to echo a new speed into the fan*_target sysfs file, and the fan speeds changed, however the new target was not reflected back in the sysfs file. Only a -1 is always displayed.

After checking a few other systems, this is evident on all with atleast r19, so it may have been introduced sooner.

msbarth, 2017-12-13:
@amboar Was there changes possibly integrated in recently for the max31785 driver?

amboar, 2017-12-16:
> VERSION_ID="ibm-v2.0-0-r19-0-ge7f2422-dirty"
Please provide the output of `uname -a` and a link to the associated kernel tree that was running on the machine, with all relevant patches included.

msbarth, 2017-12-18:
Another system without a `dirty` build and no patches:
```
root@z2632w41:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v2.0"
VERSION_ID="ibm-v2.0-0-r19-0-ge7f2422"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v2.0"
BUILD_ID="ibm-v2.0-0-r19"
root@z2632w41:~# cd /sys/class/hwmon/hwmon9/
root@z2632w41:/sys/class/hwmon/hwmon9# cat fan*_target
-1
-1
-1
-1
root@z2632w41:/sys/class/hwmon/hwmon9# uname -a
Linux z2632w41 4.10.17-e7ae4c9b5df12838e66a25df93df02e47bdb7d11 #1 Fri Dec 8 13:28:54 CST 2017 armv6l GNU/Linux
```
This seems to be pervasive since the r18 build timeframe

geissonator, 2017-12-18:
The current theory is this issue is related to us pulling in http://patchwork.ozlabs.org/patch/839887/.  We're going to pull it out of our next internal release to see if it helps.

amboar, 2017-12-19:
I've sent a v2 of the patch that introduced the bug, as it was never applied to dev-4.10:

http://patchwork.ozlabs.org/patch/850801/

I included a description of what went wrong, which is quoted below:

```
Issue 2715 was a result of type mismatch between the {read,write}_word_data()
callbacks and the pmbus_{read,write}_word_data() helper functions, where the
latter takes u8s as arguments whilst the former ints for page and reg. The
combined issues of PMBus virtual registers using values exceeding 0x100 and
incomplete handling of necessary registers in the
max31785_{read,write}_word_data() implementations lead to out-of-range register
addresses being passed to pmbus_read_word_data(). Such reads will return values
with all bits set, which when interpreted as a signed value gives -1.
```

rfrandse, 2018-03-20:
==== State: Verify by: gkeishin on 16 February 2018 00:08:35 ==== 24
Had a word with Andrew Jeffery about the test result here manually ran and looked good. So we conclude his fix worked well here.

closed 2018-03-20 by rfrandse