openbmc/openbmc #2719: GPU fault polling of 1s is too slow to reliably fault isolate GPU thermal faults

cdeadmin, 2017-12-14:


spinler, 2017-12-14:
On Witherspoon only, we need the following 2 changes:

1) Poll the GPU Overtemp signal on the UCD90160 every 500ms instead of 1s, to increase the likelihood we can catch the overtemp before it converts into a PGOOD fault (after 1s of overtemp, the GPU will shut off).

2) Poll the master PGOOD SYS_PWR_OK every 500ms instead of 1s, so we will catch other faults faster and power the system off in less time to help prevent possible damage.

rfrandse, 2017-12-14:
<https://gerrit.openbmc-project.xyz/8309> Change wspoon GPU overtemp polling to 500ms
Resolves  #2719 GPU fault polling of 1s is too slow to reliably fault isolate GPU thermal faults

closed 2017-12-23 by geissonator