openbmc/openbmc #2720: kernel prints logged to the journal twice

geissonator, 2017-12-14:
There's gotta be some config option somewhere to make it so we only log kernel print messages to the journal once.  I think maybe we need to tell syslog to not forward it on or something?  This story is to investigate into this.  Our journal space is precious so we'd like to only see kernel messages logged once in it:

```
Dec 14 20:19:05 witherspoon kernel: ftgmac100 1e660000.ethernet eth0: NCSI: HNCDSC AEN - channel 0 state down
Dec 14 20:19:05 witherspoon kernel: ftgmac100 1e660000.ethernet eth0: NCSI: suspending channel 0
Dec 14 20:19:05 witherspoon kernel: ftgmac100 1e660000.ethernet eth0: NCSI: configuring channel 0
Dec 14 20:19:05 witherspoon kernel: ftgmac100 1e660000.ethernet eth0: no vlan ids left to set
Dec 14 20:19:05 witherspoon kernel: ftgmac100 1e660000.ethernet eth0: NCSI: channel 0 config done
Dec 14 20:19:05 witherspoon kernel: ftgmac100 1e660000.ethernet eth0: NCSI: No more channels to process
Dec 14 20:19:05 witherspoon kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface up
Dec 14 20:19:05 witherspoon kernel[978]: [  589.130000] ftgmac100 1e660000.ethernet eth0: NCSI: HNCDSC AEN - channel 0 state down
Dec 14 20:19:05 witherspoon kernel[978]: [  589.130000] ftgmac100 1e660000.ethernet eth0: NCSI: suspending channel 0
Dec 14 20:19:05 witherspoon kernel[978]: [  589.130000] ftgmac100 1e660000.ethernet eth0: NCSI: configuring channel 0
Dec 14 20:19:05 witherspoon kernel[978]: [  589.130000] ftgmac100 1e660000.ethernet eth0: no vlan ids left to set
Dec 14 20:19:05 witherspoon kernel[978]: [  589.130000] ftgmac100 1e660000.ethernet eth0: NCSI: channel 0 config done
Dec 14 20:19:05 witherspoon kernel[978]: [  589.130000] ftgmac100 1e660000.ethernet eth0: NCSI: No more channels to process
Dec 14 20:19:05 witherspoon kernel[978]: [  589.130000] ftgmac100 1e660000.ethernet eth0: NCSI interface up

```

BenjaminFair, 2018-06-12:
The duplicate messages are coming from busybox klogd through syslog. We can either disable klogd (meaning syslog won't have kernel messages in it) or tell journald to stop reading kmsg.

geissonator, 2018-06-29:
Thanks for the tip @BenjaminFair - I pushed up https://gerrit.openbmc-project.xyz/#/c/11358/ which seems to get us down to just 1 for each kernel message in the journal.

rfrandse, 2018-07-02:
<https://gerrit.openbmc-project.xyz/11358> Prevent duplicate kernel message in journal
Resolves  #2720 kernel prints logged to the journal twice

closed 2018-07-24 by geissonator