openbmc/openbmc #2724: BMC won't boot to Standby Ready

gkeishin, 2017-12-18:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.0-137"
VERSION_ID="v2.0-137-g4f1dc5a"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.0-137"
BUILD_ID="v2.0"
root@witherspoon:~# systemctl list-jobs --no-pager
JOB UNIT                                                                           TYPE  STATE
 87 mapper-wait@-xyz-openbmc_project-inventory-system-chassis-powersupply0.service start running
 89 witherspoon-power-supply-monitor@1.service                                     start waiting
 85 witherspoon-power-supply-monitor@0.service                                     start waiting
162 obmc-led-group-start@bmc_booted.service                                        start waiting
 75 obmc-standby.target                                                            start waiting
 90 mapper-wait@-xyz-openbmc_project-inventory-system-chassis-powersupply1.service start running

6 jobs listed.
root@witherspoon:~# obmcutil state
CurrentBMCState     : xyz.openbmc_project.State.BMC.BMCState.NotReady
CurrentPowerState   : xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState    : xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:~#
```

gkeishin, 2017-12-18:
The previous version boots good to standby
```
root@witherspoon:~# obmcutil state
CurrentBMCState     : xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState   : xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState    : xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.0-135"
VERSION_ID="v2.0-135-g5391c82"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.0-135"
BUILD_ID="v2.0"
root@witherspoon:~#
```

gkeishin, 2017-12-18:
Looking at the commit, the delta we see is this
```

   Changes
    phosphor-mapper.bb: bump version (commit: 121e305) (detail)
    phosphor-hwmon.bb: bump version (commit: 4f1dc5a) (detail)
```

gkeishin, 2017-12-18:
@geissonator @bradbishop ^^^

geissonator, 2017-12-18:
I took the latest master, v2.0-138-g61832e9, and it worked ok for me.  So either the change in 138 fixed the problem (https://github.com/openbmc/openbmc/commit/61832e963b7b897b4fd13ac18f681ef27f2477e2) which doesn't seem likely or we have some sort of intermittent issue here.  

The captured FFDC has this fail initially:
```
Dec 18 06:52:35 witherspoon systemd[1]: mapper-wait@-xyz-openbmc_project-inventory-system-chassis-powersupply0.service: Main process exited, code=killed, status=15/TERM
Dec 18 06:52:35 witherspoon systemd[1]: Stopped Wait for /xyz/openbmc_project/inventory/system/chassis/powersupply0.
Dec 18 06:52:35 witherspoon systemd[1]: mapper-wait@-xyz-openbmc_project-inventory-system-chassis-powersupply0.service: Unit entered failed state.
Dec 18 06:52:35 witherspoon systemd[1]: mapper-wait@-xyz-openbmc_project-inventory-system-chassis-powersupply0.service: Failed with result 'signal'.
Dec 18 06:52:35 witherspoon systemd[1]: mapper-wait@-xyz-openbmc_project-inventory-system-chassis-powersupply1.service: Main process exited, code=killed, status=15/TERM
Dec 18 06:52:35 witherspoon systemd[1]: Stopped Wait for /xyz/openbmc_project/inventory/system/chassis/powersupply1.
Dec 18 06:52:35 witherspoon systemd[1]: mapper-wait@-xyz-openbmc_project-inventory-system-chassis-powersupply1.service: Unit entered failed state.
```

But subsequent reboots (and the data captured in the fail state) have this:
```
Dec 18 06:54:31 witherspoon phosphor-image-updater[1157]: Error in mapper GetSubTreePath
```

gkeishin, 2017-12-18:
Retry and it worked this time but saw the journald crashed out for some reason?

```
root@witherspoon:~# systemctl status systemd-journald.service | cat
Warning: systemd-journald.service changed on disk. Run 'systemctl daemon-reload' to reload units.
● systemd-journald.service - Journal Service
   Loaded: loaded (/lib/systemd/system/systemd-journald.service; static; vendor preset: enabled)
  Drop-In: /lib/systemd/system/systemd-journald.service.d
           └─systemd-journald-override.conf
   Active: active (running) since Fri 2017-12-15 03:20:13 UTC; 4min 47s ago
     Docs: man:systemd-journald.service(8)
           man:journald.conf(5)
 Main PID: 956 (systemd-journal)
   Status: "Processing requests..."
   CGroup: /system.slice/systemd-journald.service
           └─956 /lib/systemd/systemd-journald

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
root@witherspoon:~#
```

geissonator, 2017-12-18:
That last one was the whole "time going backwards" issue.  

I'm not sure if it's related but we did find that master was broken due to a recent mapper change.  For now we've merged a revert in https://gerrit.openbmc-project.xyz/#/c/8337/

closed 2017-12-18 by geissonator