openbmc/openbmc #2725: Witherspoon: Initiate dump when host is checkstopped

cdeadmin, 2017-12-18:


geissonator, 2017-12-18:
The host team has requested we generate a BMC dump when a checkstop occurs.  The main reason for this is that our dump will collect the host console log which is very useful for debugging the different checkstops.  This issue will be used to add the host console to the error based BMC dumps and to collect this BMC dump when the org.open_power.Host.Boot.Error.Checkstop error is created.

rfrandse, 2017-12-27:
<https://gerrit.openbmc-project.xyz/8349> Enable obmcconsole script for elog type error
Resolves  #2725 System: Initiate dump when host is checkstopped

ojayanth, 2018-01-05:
Need to redesign the code to address the review comments from @bradbishop

rfrandse, 2018-09-04:
moved to https://github.com/ibm-openbmc/dev/issues/150

closed 2018-09-04 by rfrandse