openbmc/openbmc #2727: STC:OP910:witherspoon:psu0/1_output_voltage showing as 390V instead of 12V

cdeadmin, 2017-12-18:


bjwyman, 2017-12-18:
When reading the power supply output voltage over REST, on a Witherspoon system with new power supply firmware revisions, the Value is around 390,000 millivolts instead of the expected 12,000 millivolts.

bjwyman, 2017-12-18:
```
$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://$BMC_IP/xyz/openbmc_project/sensors/voltage/ps1_output_voltage                                                                                                                                                            
{                                                                                                                                                             
  "data": {                                                                                                                                                   
    "CriticalAlarmHigh": 1,                                                                                                                                   
    "CriticalAlarmLow": 0,                                                                                                                                    
    "CriticalHigh": 13000,                                                                                                                                    
    "CriticalLow": 0,                                                                                                                                         
    "Scale": -3,                                                                                                                                              
    "Unit": "xyz.openbmc_project.Sensor.Value.Unit.Volts",                                                                                                    
    "Value": 389000,                                                                                                                                          
    "WarningAlarmHigh": 1,                                                                                                                                    
    "WarningAlarmLow": 0,                                                                                                                                     
    "WarningHigh": 12700,                                                                                                                                     
    "WarningLow": 0                                                                                                                                           
  },                                                                                                                                                          
  "message": "200 OK",                                                                                                                                        
  "status": "ok"
$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://$BMC_IP/xyz/openbmc_project/sensors/voltage/ps1_output_voltage                                                                                                                                                            
{                                                                                                                                                             
  "data": {                                                                                                                                                   
    "CriticalAlarmHigh": 1,                                                                                                                                   
    "CriticalAlarmLow": 0,                                                                                                                                    
    "CriticalHigh": 13000,                                                                                                                                    
    "CriticalLow": 0,                                                                                                                                         
    "Scale": -3,                                                                                                                                              
    "Unit": "xyz.openbmc_project.Sensor.Value.Unit.Volts",                                                                                                    
    "Value": 389000,                                                                                                                                          
    "WarningAlarmHigh": 1,                                                                                                                                    
    "WarningAlarmLow": 0,                                                                                                                                     
    "WarningHigh": 12700,                                                                                                                                     
    "WarningLow": 0                                                                                                                                           
  },                                                                                                                                                          
  "message": "200 OK",                                                                                                                                        
  "status": "ok"
```

bjwyman, 2017-12-18:
There is a conversion factor that needs to be used to calculate the millivolts value from the value we are reading from in2_input.

hwmon allows for this via gain and offset lines in the configuration file.

rfrandse, 2017-12-19:
<https://gerrit.openbmc-project.xyz/8342> Add GAIN line for power supply output voltage
Resolves:  #2727 System:psu0/1_output_voltage showing as 390V instead of 12V

bjwyman, 2018-01-03:
The commit I pushed up earlier should probably be abandoned. This is looking more and more like a device driver update. I cannot comment on the code up in Gerrit right now, "Server Error".

bjwyman, 2018-01-05:
Cherry-picking a fix from dev-4.13 to dev-4.10 seemed to resolve the issue on a system with the latest power supply firmware:
https://github.com/openbmc/linux/commit/e608cab9f66c0bcc633d63542cfb5ad1374f4acb
However, power supplies with older firmware do not appear to work with just that change.

bjwyman, 2018-01-05:
http://patchwork.ozlabs.org/patch/856269/

bjwyman, 2018-01-09:
hwmon (pmbus): cffps: Add PMBUS_SKIP_STATUS_CHECK
https://patchwork.kernel.org/patch/10150591/

bjwyman, 2018-01-12:
https://gerrit.openbmc-project.xyz/8468

bjwyman, 2018-01-16:
The 8468 commit noted earlier was flagged as -1, as the changes should get merged to the Linux dev-4.10 repo. Those look to be merged now.
https://github.com/openbmc/linux/commit/3339bdef88d3d546bca653c6e6fa101283c252ff
https://github.com/openbmc/linux/commit/dd710e71ee53b95f4cdacdd364b648263aea251a

Master should have those as soon as the SRCREV in the openbmc is bumped.

bjwyman, 2018-01-17:
Kernel bump that pulls this into master:
https://github.com/openbmc/openbmc/commit/dda4bff84fb051a5da9827ab6d61a4e38ce2d64f

rfrandse, 2018-01-30:
<https://gerrit.openbmc-project.xyz/8468> kernel: patch in pmbus and power supply driver fixes
Resolves  #2727 System:psu0/1_output_voltage showing as 390V instead of 12V

rfrandse, 2018-01-30:
Both of those changes are in master, in Linux, via the autobump Brandon mentioned in the issue.

closed 2018-01-30 by rfrandse