openbmc/openbmc #2728: Set time failed via date

KenChenIEC, 2017-12-19:
Hi all,
I try to set BMC time and get second, reset the time by second. 
The time is wired.

Here is test result:
root@2400-rep:~# date -s "2017-12-12 01:01:01"
Tue Dec 12 01:01:01 UTC 2017
root@2400-rep:~# date +%s
1513040463
root@2400-rep:~# date -s 1513040463
Mon Jan  4 05:03:00 UTC 2016
root@2400-rep:~# date +%s
1451883786

chiabinglee, 2017-12-19:
Hi Ken,
Please try "timedatectl" command, and must disable ntp at first.
```
root@:~# timedatectl
      Local time: Tue 2017-12-19 09:49:15 UTC
  Universal time: Tue 2017-12-19 09:49:15 UTC
        RTC time: Tue 2017-12-19 09:49:16
       Time zone: n/a (UTC, +0000)
 Network time on: yes
NTP synchronized: yes
 RTC in local TZ: no
root@:~# timedatectl set-ntp false
Dec 19 09:51:59  dbus[835]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service'
Dec 19 09:51:59  systemd[1]: Starting Time & Date Service...
Dec 19 09:51:59  dbus[835]: [system] Successfully activated service 'org.freedesktop.timedate1'
Dec 19 09:51:59  systemd[1]: Started Time & Date Service.
Dec 19 09:51:59  systemd-timedated[2541]: Failed to get target of /etc/localtime: No such file or directory
Dec 19 09:51:59  systemd[1]: Reloading.
root@:~# Dec 19 09:52:03  systemd[1]: Stopping Network Time Synchronization...
Dec 19 09:52:03  systemd-timedated[2541]: Set NTP to disabled
Dec 19 09:52:03  systemd[1]: Stopped Network Time Synchronization.

root@:~# timedatectl set-time "2017-02-17 16:33:00"
root@:~# Feb 17 16:33:00  phosphor-timemanager[1033]: BMC system time is changed
Feb 17 16:33:00  systemd[1]: Time has been changed
Feb 17 16:33:00  systemd-timedated[2541]: Changed local time to Fri Feb 17 16:33:00 2017
timedatectl
      Local time: Fri 2017-02-17 16:33:09 UTC
  Universal time: Fri 2017-02-17 16:33:09 UTC
        RTC time: Fri 2017-02-17 16:33:09
       Time zone: n/a (UTC, +0000)
 Network time on: no
NTP synchronized: no
 RTC in local TZ: no
```

KenChenIEC, 2017-12-20:
Hi @chiabinglee,
Thanks, the result is right with your step.
But my platform of BMC get time from the host and the time unit is second. 
Has the set time support unit is second?
or I have to transfer the second to fully time format?

mine260309, 2017-12-20:
@KenChenIEC The `date` tool in busybox is not the same as GNU `date`, the supported format is:
```
        hh:mm[:ss]
        [YYYY.]MM.DD-hh:mm[:ss]
        YYYY-MM-DD hh:mm[:ss]
        [[[[[YY]YY]MM]DD]hh]mm[.ss]
```
See https://busybox.net/downloads/BusyBox.html

Your command `date -s 1513040463` won't work as your expectation.