openbmc/openbmc #2729: OpenBMC GUI - In the title frame IP address of the system and Name has to display properly

samahaba, 2017-12-19:
[image]

sivassrr, 2017-12-19:
@samahaba I had already opened an issue for this. 
I think it is dup of https://github.com/openbmc/openbmc/issues/2648

closed 2017-12-19 by sivassrr

samahaba, 2017-12-19:
see here one more issue is there that is it is showing label as "BMC IP address" but in front of this system name is displaying that is "002.aus.stglabs.ibm.com". Here system name is also not correct it should display as "wsbmc002.aus.stglabs.ibm.com" not simply as "002.XXX.XXX.com"

samahaba, 2017-12-19:
Please here am requesting you that it will be good if you discuss with me before closing the issue as like this