openbmc/openbmc #2730: OpenBMC GUI - Editing of System name is not possible

samahaba, 2017-12-19:
[image]

iftekharuli, 2018-03-13:
@ratagupt I must have asked you this before- what is the rest command to rename the system name?

ratagupt, 2018-03-14:
@iftekharuli I thought I had given it before in the mail which I sent it few months back :)
Anyways you can refer the section("Configure HostName") in the following link.
https://gerrit.openbmc-project.xyz/#/c/6872/5/xyz/openbmc_project/Network/README.md

gtmills, 2018-04-28:
same issue here: https://github.com/openbmc/openbmc/issues/3104

gtmills, 2018-04-30:
Going to close this issue since is the same issue as #3104. Typically I would close the newer one and dub to the older one, but the newer has a TODO in the code.

closed 2018-04-30 by gtmills