openbmc/openbmc #2732: Gard clear API vs opal-gard clear restore back original GUARD record

gkeishin, 2017-12-19:
After discussing with @gtmills and @mtritz 
API clear is working as intended.

It’s known that those two things are behaving differently, but there might be some design change to the gard clear API.

```
root@witherspoon:~# ls -larth /var/lib/phosphor-software-manager/pnor/prsv/GUARD
-rw-rw-r--    1 root     root       20.0K Dec 18 22:02 /var/lib/phosphor-software-manager/pnor/prsv/GUARD
root@witherspoon:~# 
```
* Clear the record:
```
URI:https://xx.xx.xx.xx/org/open_power/control/gard/action/Reset, method:Post, args:{u'data': {u'data': []}, u'headers': {u'Content-Type': u'application/json'}}
Response code:200, Content:{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
```
```
root@witherspoon:~# ls -larth /var/lib/phosphor-software-manager/pnor/prsv/GUARD
ls: /var/lib/phosphor-software-manager/pnor/prsv/GUARD: No such file or directory
root@witherspoon:~#
```

* Now after clearing, let try to clear via opal-gard via OS
```
root@rcx2d508:~# opal-gard clear all
Clearing the entire gard partition...done
root@rcx2d508:~#
```
* The Gard restored back ?
```
root@witherspoon:~# ls -larth /var/lib/phosphor-software-manager/pnor/prsv/GUARD
-rw-rw-r--    1 root     root       20.0K Dec 18 22:15 /var/lib/phosphor-software-manager/pnor/prsv/GUARD
root@witherspoon:~#
```

gkeishin, 2017-12-19:
Starting with @anoo1 on this for inputs here

oohal, 2018-01-02:
The opal-gard too clears the GUARD partition by doing an erasing the partition then programming the ECC bits so the GUARD file will still exist afterwards. If you wanted to have consistent behaviour between opal-gard and the rest API then you will need to add something to mboxd to detect and remove GUARD when it's empty. A clean-ish way to do it would be to compare the contents of the pnor-prsv version and the pnor-ro version and discard the prsv file if it's identical.

amboar, 2018-04-18:
Why is it necessary to have consistency?

amboar, 2019-04-02:
Closing this as there's no requirement for "fixing" the behaviour.

closed 2019-04-02 by amboar