openbmc/openbmc #2733: occ's not going active with master

geissonator, 2017-12-19:
Not sure yet if this is specific to my system (w54) or if it's pervasive in master but the occ's are not both going to active state when running the latest code out of master on my witherspoon.

```
Dec 19 16:55:16 witherspoon ipmid[1249]: ERROR:[0x80] handling NetFn:[0x2C], Cmd:[0x3]
Dec 19 16:55:16 witherspoon ipmid[1249]: Bus: :1.68, Path: /org/open_power/control/occ1, Interface: org.open_power.OCC.Status
Dec 19 16:55:18 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Dec 19 16:55:19 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-sbefifo\x402400-occ\x401-occ\x2dhwmon\x401.service: Unit not needed anymore. Stopping.
Dec 19 16:55:19 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
Dec 19 16:55:19 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
```

```
v2.0-142-g05e02d5
```

geissonator, 2017-12-19:
I added some extra trace.  It looks like the old problem where the openpower-occ-control application starts looking for occ errors before both devices are active.
```
Dec 19 20:58:34 witherspoon ipmid[1249]: Bus: :1.286, Path: /org/open_power/control/occ1, Interface: org.open_power.OCC.Status
Dec 19 20:58:35 witherspoon openpower-occ-control[4377]: Setting occ active property
-> OCC 0, ACTIVE True
Dec 19 20:58:36 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Dec 19 20:58:37 witherspoon openpower-occ-control[4377]: StatusCallBack: true
Dec 19 20:58:37 witherspoon openpower-occ-control[4377]: OCC Error Detected
-> OCC 0
Dec 19 20:58:37 witherspoon openpower-occ-control[4377]: Setting occ active property
-> OCC 0, ACTIVE False
Dec 19 20:58:37 witherspoon openpower-occ-control[4377]: StatusCallBack: false
Dec 19 20:58:37 witherspoon ipmid[1249]: Pushing cmd on to queue
Dec 19 20:58:37 witherspoon ipmid[1249]: Pushing cmd on to queue
Dec 19 20:58:37 witherspoon ipmid[1249]: Asserting SMS Attention
-> BMC code requesting a BMC reboot to the OS
Dec 19 20:58:37 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-sbefifo\x402400-occ\x401-occ\x2dhwmon\x401.service: Unit not needed anymore. Stopping.
Dec 19 20:58:37 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Dec 19 20:58:37 witherspoon ipmid[1249]: SMS Attention asserted
Dec 19 20:58:37 witherspoon openpower-occ-control[4377]: Setting occ active property
-> OCC 1, ACTIVE True
Dec 19 20:58:38 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Dec 19 20:58:38 witherspoon openpower-occ-control[4377]: StatusCallBack: true
Dec 19 20:58:38 witherspoon openpower-occ-control[4377]: OCC Error Detected
-> OCC 1
Dec 19 20:58:39 witherspoon phosphor-watchdog[4384]: watchdog: disabled
Dec 19 20:58:40 witherspoon ipmid[1249]: Host control timeout hit!
Dec 19 20:58:40 witherspoon ipmid[1249]: Timer expired
Dec 19 20:58:40 witherspoon openpower-occ-control[4377]: Host control signal values
Dec 19 20:58:40 witherspoon openpower-occ-control[4377]: Error resetting the OCC.
Dec 19 20:58:40 witherspoon openpower-occ-control[4377]: Host control signal values
Dec 19 20:58:40 witherspoon openpower-occ-control[4377]: Error resetting the OCC.
```

geissonator, 2017-12-19:
If we activate occ 1 first, and then activate occ 0, all works fine.  But when we activate occ 0 first, we hit the above error.

mine260309, 2017-12-20:
Reproduced on Romulus as well.

I guess something wrong with error watch in `openpower-occ-control`, which is enabled in https://github.com/openbmc/openpower-occ-control/commit/9fd2bdcab930620e67fce69dc6a9e6e11e9880b4

geissonator, 2017-12-20:
Eddie is taking a look at this now

eddiejames, 2017-12-20:
Looks like the presence watching for OCCs is hitting somewhat of a race condition between the OCC reporting both OCCs and both OCCs being bound. We just need to wait for both OCCs bound before starting the presence watching.

rfrandse, 2017-12-21:
<https://gerrit.openbmc-project.xyz/8358> Only add presence watching when all OCCs are bound
Resolves  #2733 occ's not going active with master

closed 2018-01-04 by geissonator