openbmc/openbmc #2735: OpenBMC GUI - Power Consumption and Power Cap Info is not correct

samahaba, 2017-12-20:
[image]

gtmills, 2018-02-08:
https://github.com/openbmc/phosphor-webui/blob/db28a387c218b82ff70766359f4a521c713ab274/app/overview/controllers/system-overview-controller.html#L74
Power consumption and Power Cap Info are always set to 0. 
We will need to make a rest call here:

For the power_cap 
 curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET  https://${BMC_IP}/xyzopenbmc_project/control/host0/power_cap
{
  "data": {
    "PowerCap": 0, 
    "PowerCapEnable": 0
  }, 
  "message": "200 OK", 
  "status": "ok"

2nd example
{
    "PowerCap": 1750,
    "PowerCapEnable": 1
}

We should display "Not Enabled" if "PowerCapEnable" = 0. If "PowerCapEnable" = 1 then we should display the "PowerCap" value (e.g. 1750 in the 2nd example)

rfrandse, 2018-03-14:
<https://gerrit.openbmc-project.xyz/9554> Display correct Power Consumption & Cap values
Resolves  #2735 OpenBMC GUI - Power Consumption and Power Cap Info is not correct

closed 2018-04-13 by geissonator