openbmc/openbmc #2736: OpenBMC GUI - Serial Over LAN functionality is not working

samahaba, 2017-12-20:
[image]

gtmills, 2018-03-01:
To get the host console/SOL, ssh to port 2200. 
Googling, it looks to be Serial Over LAN (SOL) not "SoL".

jaypadath, 2018-03-15:
Options tried so far:
1) With hterm + nassh. I could launch it from a separate window and ssh to host. But this has a drawback for supporting only as a chrome extension.
2) WebSSH2. This uses xterm.js. Able to get into ssh through "http://localhost:2222/ssh/host/wsbmc019.aus.stglabs.ibm.com" after installing in to my system. Need to check further on the feasibility of integrating into our package.
3) Wetty which uses hterm+ssh
I had installed the wetty in my system. With this by specifying the sshhost (say wsbmc019.aus.stglabs.ibm.com) and port (2200), I am able reach to the ssh console (localhost:3000). As of now this supports the command line option ie to start like "node app.js --sshhost wsbmc019.aus.stglabs.ibm.com --sshuser root --sshport 2200 -p 3000".
Not able to find a right way to merge this in to our npm for Web GUI. 
4) https://github.com/butlerx/wetty , this is a forked version which talks building wetty as a node_module. But I am getting some build error while installing it in my system.
Also the forked version uses xterm instead of hterm. This one talks about starting wetty as service with systemd.

jaypadath, 2018-03-19:
Options tried/analysed so far to enable SOL with ssh in openbmc webgui:

1) With hterm + nassh. I could launch it from a separate window and ssh to host. But this has a drawback for supporting only as a chrome extension.
2) WebSSH2. This uses xterm.js. Able to get into ssh through "http://localhost:2222/ssh/host/wsbmc019.aus.stglabs.ibm.com" after installing in to my system. Need to check further on the feasibility of integrating into our package. Also not yet received any response/help from community.
3) Wetty which uses hterm+ssh : https://github.com/krishnasrinivas/wetty
   I had installed the wetty in my system. With this by specifying the sshhost (say wsbmc019.aus.stglabs.ibm.com) and port (2200), I am able reach to the ssh console 
   (localhost:3000). As of now this supports the command line option ie to start like "node app.js --sshhost wsbmc019.aus.stglabs.ibm.com --sshuser root --sshport 2200 -p 3000".
   Not able to find a right way to merge this in to our npm for Web GUI.
4) https://github.com/butlerx/wetty , this is a forked version which talks building wetty as a node_module. This forked version uses xterm instead of hterm. This one also talks
   about starting wetty as service with systemd. Solved the issue of installation in my system. SSH to localhost is working but ssh to witherspoon was not working.
   Got response from the community. Tried to do some steps suggested by butlerx. But that change alone did not do anything. Further tried to install some dependent package and
   it triggered the corruption of web gui package.
5) https://github.com/nickola/web-console
   Looks to be more proprietary and PHP based one. Could not make it work as some data needed to be uploaded to web which I did not try.
6) https://github.com/R3nPi2/angular-ssh
   Looks to be an easy one with angular-ssh client. But have dependency on cordova-plugin-sshclient which work only on Android platform.
7) https://github.com/vtortola/ng-terminal-emulator
   Nice terminal emulator completely written in angular js. Was facing some issue to port to our code. This code does not support the ssh functionality internally.
8) https://gist.github.com/zpiman/254156a211479f7ee240
   This is a simple terminal emulator which uses packages like express, socket.io and pty.js. I have tried to integrate this with our project. But while installing pty.js, our
   node.js package is getting corrupted and could not test it due to that.

gtmills, 2018-05-03:
https://gerrit.openbmc-project.xyz/#/c/10271/ fixed this. Closing.

closed 2018-05-03 by gtmills