openbmc/openbmc #2740: IPMI: undefined symbol _ZTINSt13__future_base12_Result_baseE

artemsen, 2017-12-21:
It seems like commit [61832e](https://github.com/openbmc/openbmc/commit/61832e963b7b897b4fd13ac18f681ef27f2477e2) (Add support for c++ futures for armv5te) doesn't work properly with the latest phoshor-host-ipmid.

Library libapphandler.so cannot be loaded by IPMI daemon:
`Dec 21 13:52:01 manzoni ipmid[995]: ERROR opening [/usr/lib/host-ipmid/libapphandler.so.0]: /usr/lib/host-ipmid/libapphandler.so.0: undefined symbol: _ZTINSt13__future_base12_Result_baseE`

Symbol `_ZTINSt13__future_base12_Result_baseE` is typeinfo for `std::__future_base::_Result_base`. It was added to phoshor-host-ipmid with commit [5bfbde](https://github.com/openbmc/phosphor-host-ipmid/commit/5bfbde351aabe2096de99ee959a8a2e50edcf84c).

feistjj, 2018-01-04:
What platform are you building for? Is it using an ast2400?

artemsen, 2018-01-04:
Yes, it is ast2400.

feistjj, 2018-01-04:
Which platform?

artemsen, 2018-01-04:
OpenPOWER, custom modification from palmetto config without any custom patches on GCC.

spg2, 2018-01-10:
We are facing same issue.
Jan 09 09:49:52 palmetto ipmid[923]: ERROR opening [/usr/lib/host-ipmid/libapphandler.so.0]: /usr/lib/host-ipmid/libapphandler.so.0: undefined symbol: _ZTINSt13__future_base12_Result_baseE

feistjj, 2018-01-10:
We probably should just revert the changes until we update to yocto 2.4 which has the newer version of GCC with the fix for armv5 futures as the backport seemed to fail.

mine260309, 2018-02-07:
I am running latest OpenBMC on Palmetto and hit this issue.

@bradbishop Is there any date about when we can move to yocto 2.4?

This issue prevent all P8 systems (with ast2400) booting with latest OpenBMC, because `ipmid` fails to load `libapphandler.so` and cause watchdog will not be kicked and always timeout on power on.

shenki, 2018-02-09:
I am also being hit by this on Palmetto in my efforts to move to Linux dev-4.10.

rfrandse, 2018-02-09:
<https://gerrit.openbmc-project.xyz/8917> meta-phosphor: ipmi-host: revert identify IPMI command
Resolves  #2740 IPMI: undefined symbol _ZTINSt13__future_base12_Result_baseE

shenki, 2018-02-14:
Why has this been assigned to me?

This is the responsibility of those working on ipmid.

shenki, 2018-02-26:
No one has done anything, so I sent a revert:

 https://gerrit.openbmc-project.xyz/#/c/9210/

shenki, 2018-03-01:
Fixed in master as of 32004fa5039c574b254b63af363b7d16592a2106

closed 2018-03-01 by shenki