openbmc/openbmc #2741: OpenBMC GUI - "Users" main menu -> "Manage user account" Sub menu is not wokring

samahaba, 2017-12-22:
[image]

samahaba, 2017-12-22:
"Manage user account" sub menu functionality is not working as per the latest OpenBMC gui code dated 22nd December 2017

samahaba, 2017-12-22:
"Manage user account" sub menu functionality is not working as per the latest OpenBMC gui code dated 22nd December 2017. phosphor-webui-master code repo

iftekharuli, 2018-03-14:
We did not code this part due to the lack of the completion rest command when we handed off the GUI. This is being resolved now in https://github.com/openbmc/openbmc/issues/2741

gtmills, 2018-03-20:
https://gerrit.openbmc-project.xyz/#/c/9610 resolves this

rfrandse, 2018-03-20:
<https://gerrit.openbmc-project.xyz/9610> Display error and success text
Resolves  #2741 OpenBMC GUI - "Users" main menu -> "Manage user account" Sub menu is not working

closed 2018-03-23 by geissonator