openbmc/openbmc #2742: qemu-system-arm palmetto error

lockless, 2017-12-26:
I try to emulate palmetto, when type:

    root@Ubuntu:/home/skunlin/qemu/build/arm-softmmu# ./qemu-system-arm -m 256 -M palmetto-bmc -nographic -drive file=./flash-palmetto,format=raw,if=mtd -net nic -net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostname=qemu

the screen output follow information, then the cpu 100%, and can't continue anymore.

    qemu-system-arm: Aspeed iBT has no chardev backend

    DRAM Init-DDR3
    CBR0-1

when try ssh the ,it seems no response.

    root@Ubuntu:/home/skunlin/qemu/build/arm-softmmu# curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://localhost:2443/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"

can i get some advise?

bgrupczy, 2017-12-26:
> I try to emulate palmetto, when type:
>
> root@Ubuntu:/home/skunlin/qemu/build/arm-softmmu# ./qemu-system-arm -m 256 -M palmetto-bmc -nographic -drive file=./flash-palmetto,format=raw,if=mtd -net nic [...]

I had the same issue with my first build. I did a clean pull, rebuilt, and it started working. It’s the only “solution” I have to offer. You may want to keep your existing image for someone to look at. 

It has not booted up so you will not be able to ssh or anything else with it. 

-Brad

lockless, 2017-12-26:
I suppose the build process has no problem.
When using the "flash-palmetto" download from autobuild "https://openpower.xyz/", it has the same output.

closed 2019-04-27 by stale[bot]