openbmc/openbmc #2743: OpenBMC GUI - Sub menu is not hiding or closing

samahaba, 2017-12-26:
[image]

samahaba, 2017-12-26:
Problem Description.
Open any sub menu and click any ware other than menu and sub menu area the sub menu should hide/close but here it is not hide/close.

sivassrr, 2018-01-23:
@samahaba Do you want header to get close? Header will be always there to display the current status about the system. Below the header, frames applicable to that will get change. 

Or, You are asking about the frame itself. Getting confuse with your arrow. May be you want to highlight with right circle/  rectangle / square box.

samahaba, 2018-01-23:
If I click some are on the gui is should hide.. If you want I will show you the demo that what is expectation if  the GUI is more user friendly from the user prospective.

rfrandse, 2018-02-08:
<https://gerrit.openbmc-project.xyz/8915> Fixing the submenu closing issue
fixes  #2743 OpenBMC GUI - Sub menu is not hiding or closing

closed 2018-03-14 by geissonator