openbmc/openbmc #2744: OpenBMC GUI - "Server health" -> "Event log", option "Medium" is not able to select independently

samahaba, 2017-12-26:
[image]

samahaba, 2017-12-26:
Problem Description
-----------------------

1. Go to  "Server health" -> "Event log"
2. Click button "High" and then click "Medium" button. here indicate/looks that both "High" and "Medium" buttons are selected, but "High" button has to deselect after clicking on "Medium" button.

iftekharuli, 2018-02-01:
yes. You are correct. That is how we wanted to design. The idea was to be able to select multiple events together. Meaning- lets say someone wants to see both high and medium events, they click on both- high and medium events are shown. If they want to see medium events only. they have to deselect high events like you mentioned by clicking on high again.

Let me know if you have any more questions, thanks!

iftekharuli, 2018-03-15:
@samahaba this is similar to #2750 .. I have replied in detail about the behavior. Let me know if you need more info

gtmills, 2018-03-15:
The GUI was designed this way. Closing this issue.

closed 2018-03-15 by gtmills