openbmc/openbmc #2745: Romulus occ hwmon failure on host power on

mine260309, 2017-12-28:
On Romulus, occ hwmon always fails when host is powered on from power off.

The related log is extracted happens on host sets occ active sensor to true.
Some observation:
1. First it gets -74 (`EBADMSG`) error;
2. Then it gets -5 (`EIO`) error;
3. openpower-occ-control get the error, and tries to reset occ;

```
Dec 28 02:27:40 rom106 kernel: occ-hwmon occ1-dev0: occ bus op failed rc:-74
Dec 28 02:27:40 rom106 kernel[974]: [72844.670000] occ-hwmon occ1-dev0: occ bus op failed rc:-74
Dec 28 02:27:42 rom106 kernel: sbefifo 00:00:00:06: Fatal bus access failure: -5
Dec 28 02:27:42 rom106 kernel: occ-hwmon occ1-dev0: occ bus op failed rc:-5
Dec 28 02:27:42 rom106 kernel[974]: [72846.720000] sbefifo 00:00:00:06: Fatal bus access failure: -5
Dec 28 02:27:42 rom106 kernel[974]: [72846.720000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
Dec 28 02:27:45 rom106 kernel: sbefifo 01:01:00:06: Fatal bus access failure: -5
Dec 28 02:27:45 rom106 kernel: occ-hwmon occ2-dev0: occ bus op failed rc:-5
Dec 28 02:27:45 rom106 kernel: occ-hwmon occ1-dev0: occ bus op failed rc:-5
Dec 28 02:27:45 rom106 kernel[974]: [72849.280000] sbefifo 01:01:00:06: Fatal bus access failure: -5
Dec 28 02:27:45 rom106 kernel[974]: [72849.280000] occ-hwmon occ2-dev0: occ bus op failed rc:-5
Dec 28 02:27:45 rom106 kernel[974]: [72849.280000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
Dec 28 02:27:45 rom106 systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-sbefifo\x402400-occ\x401-occ\x2dhwmon\x401.service: Unit not needed anymore. Stopping.
Dec 28 02:27:45 rom106 openpower-occ-control[4989]: Host control signal values
Dec 28 02:27:45 rom106 openpower-occ-control[4989]: Host control signal values
Dec 28 02:27:46 rom106 kernel: occ-hwmon occ2-dev0: occ bus op failed rc:-5
Dec 28 02:27:46 rom106 kernel[974]: [72850.540000] occ-hwmon occ2-dev0: occ bus op failed rc:-5
Dec 28 02:27:47 rom106 systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-hub\x403400-cfam\x401\x2c0-sbefifo\x402400-occ\x402-occ\x2dhwmon\x402.service: Unit not needed anymore. Stopping.
Dec 28 02:27:47 rom106 kernel: occ-hwmon occ2-dev0: occ bus op failed rc:-5
Dec 28 02:27:47 rom106 kernel[974]: [72851.730000] occ-hwmon occ2-dev0: occ bus op failed rc:-5
Dec 28 02:27:47 rom106 openpower-occ-control[4989]: Host control signal values
Dec 28 02:27:47 rom106 openpower-occ-control[4989]: Host control signal values
```

occ hwmon can be recover by manually run scan fsi and set occ active again:
```
openpower-proc-control scanFSI
occ-active.sh enable
```

mine260309, 2017-12-28:
@eddiejames 
It looks like we either need to figure out why it gets `EBADMSG` and `EIO` error and fix the root cause;
or make `openpower-occ-control` to do scanFSI and bind occ hwmon driver again?

mine260309, 2017-12-28:
Another slightly different log that the first error becomes -61 (`ENODATA`)
```
Dec 28 06:35:19 rom285 kernel: occ-hwmon occ2-dev0: occ bus op failed rc:-61
Dec 28 06:35:19 rom285 kernel[982]: [  496.150000] occ-hwmon occ2-dev0: occ bus op failed rc:-61
Dec 28 06:35:24 rom285 kernel: ftgmac100 1e660000.ethernet eth0: NCSI: LSC AEN - channel 0 state up
Dec 28 06:35:24 rom285 kernel[982]: [  501.040000] ftgmac100 1e660000.ethernet eth0: NCSI: LSC AEN - channel 0 state up
Dec 28 06:35:29 rom285 kernel: sbefifo 00:00:00:06: Fatal bus access failure: -5
Dec 28 06:35:29 rom285 kernel: occ-hwmon occ1-dev0: occ bus op failed rc:-5
Dec 28 06:35:29 rom285 kernel[982]: [  505.770000] sbefifo 00:00:00:06: Fatal bus access failure: -5
Dec 28 06:35:29 rom285 kernel[982]: [  505.770000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
Dec 28 06:35:31 rom285 kernel: occ-hwmon occ1-dev0: occ bus op failed rc:-5
Dec 28 06:35:31 rom285 kernel[982]: [  508.290000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
Dec 28 06:35:33 rom285 systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-sbefifo\x402400-occ\x401-occ\x2dhwmon\x401.service: Unit not needed anymore. Stopping.
Dec 28 06:35:33 rom285 systemd[1]: Stopping Phosphor Hwmon Poller...
Dec 28 06:35:34 rom285 systemd[1]: Stopped Phosphor Hwmon Poller.
Dec 28 06:35:34 rom285 kernel: occ-hwmon occ1-dev0: occ bus op failed rc:-5
Dec 28 06:35:34 rom285 kernel[982]: [  510.950000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
Dec 28 06:35:34 rom285 openpower-occ-control[1098]: Host control signal values
Dec 28 06:35:34 rom285 openpower-occ-control[1098]: Host control signal values
```

mine260309, 2017-12-29:
The issue only occurs when openpower-occ-control tries to bind the occ-hwmon driver at the time that it receives occ active sensor during boot.

If it bind the driver later, the issue will not occur, e.g.
1. Manually disable openpower-occ-control during boot (stop and mask `org.open_power.OCC.Control.service`)
2. Enable it after host is booted (unmask and start)
3. Then set occ active to true (`occ-active.sh enable`)

This has been confirmed by Foxconn on BarreleyeG2 and by my testing on Romulus.

So it seems to indicate that there is something wrong at the time host sets occ active sensor and BMC binds the occ-hwmon driver.
A possible workaround is to delay the bind.

eddiejames, 2018-01-11:
-5 means an fsi bus failure. Maybe some hardware issue where fsi isn't fully ready by the time we start using the second sbefifo? Not sure, we may need @cbostic help on this one.

madscientist159, 2018-01-20:
@mine260309 We've been working on laying the groundwork for a proper fix for the early OCC binding:

https://git.raptorcs.com/git/talos-skeleton/tree/pyiplobserver

This service lets anything attached to DBUS know the current IPL state., and could be used to delay OCC binding until the OCCs are active.  If you're interested in merging upstream let me know and I'll get a pull request together.

amboar, 2018-02-04:
@madscientist159 it might be worth sending mail to the list about pyiplobserver, Cc'ing @geissonator, @bradbishop and @eddiejames, mainly to give it some visibility. Without having looked at it, it seems like it could be useful to upstream.

amboar, 2018-02-05:
I more meant just bringing it up for discussion and pointing to your git repo, to start the conversation about what we might do about integrating it

madscientist159, 2018-02-05:
@ambor OK, great, thanks for the clarification!

mine260309, 2018-03-30:
With latest OpenBMC (kernel on dev-4.13), the issue is gone.
The occ/fsi driver are updated on dev-4.13 and it seems to have fixed this issue.

closed 2018-03-30 by mine260309