openbmc/openbmc #2746: OpenBMC GUI - Not able to select low priority option in event log page

samahaba, 2017-12-28:
[image]

samahaba, 2017-12-28:
Step to reproduce the problem:
------------------------------------

1 Go to "Server health" main menu -> "Event log" sub menu
2. Under "FILTER BY SEVERITY" click on option button "All", "High", "Medium", and then "Low" respectively and as result of this not able to select "Low" option

iftekharuli, 2018-02-01:
@samahaba If you select High. medium and then trying to select Low we automatically select ALL since High, medium and Low means all the events. Let me know if any questions.

iftekharuli, 2018-03-15:
@samahaba this is similar to https://github.com/openbmc/openbmc/issues/2750 .. I have replied in detail about the behavior. Let me know if you need more info

gtmills, 2018-03-15:
The GUI was designed this way. Closing this issue.

closed 2018-03-15 by gtmills