openbmc/openbmc #2747: OpenBMC GUI -> There should be appropriate and user friendly warning message has to display for "FILTER BY SEVERITY"  option

samahaba, 2017-12-28:
[image]

samahaba, 2017-12-28:
Problem description
------------------------

In "Server health" main menu -> "Event log" sub menu under "FILTER BY SEVERITY" option, if there is no "High" or "Medium" or "Low" even logs not in a list, then it will be a good have appropriate user friendly working message display

samahaba, 2017-12-28:
In "Server health" main menu -> "Event log" sub menu under "FILTER BY SEVERITY" option, if there is no "High" or "Medium" or "Low" even logs not in a list, then it will be a good have appropriate user friendly warning message display

karlvochatzer, 2018-01-05:
In addition to the "0 Events are logged"? I think you are suggesting to add a text string below this blue header that would say something like "There are no events to display that meet the filters selected." It's kind of a judgement call without usability data.

@iftekharuli Perhaps all of the Filter by Severity buttons should be disabled in the condition when there are absolutely 0 events within the system. 

[Side comment: Is there any reason why the "0 Events are logged" check box is selected by default? That should be a short cut control for the user to select all events in the log. It should not be enabled by default, rather the user should select it when s/he wants to select all of the events displayed. If this behavior, that it is selected by default is true, please copy this info into a new Github Issue.]

gtmills, 2018-08-15:
By default "0 Events are logged" checkbox should not be checked.

rfrandse, 2018-08-28:
<https://gerrit.openbmc-project.xyz/12184> Checkbox not selected with 0 log events selected
Resolves  #2747 OpenBMC GUI -> There should be appropriate and user friendly warning message has to display for "FILTER BY SEVERITY"  option

closed 2018-08-29 by geissonator