openbmc/openbmc #2748: OpenBMC GUI -> There should be appropriate and user friendly warning message has to display for "Sensors" Sub menu -> "FILTER BY SEVERITY" option

samahaba, 2017-12-28:
[image]

samahaba, 2017-12-28:
In "Server health" main menu -> "Sensors" sub menu under "FILTER BY SEVERITY" option, if there is no "Critical" or "Warning" event logs not in a list, then it will be a good have appropriate user friendly warning message display.

karlvochatzer, 2018-01-05:
Similar to #2747, I'd say that @iftekharuli should consider that the Critical is disabled when there are no sensor data results that match "critical" levels, that would then never display a null set like is shown above. Otherwise, yes a string ought to be displayed below the Sensors header that says something like "There are no sensors in a Critical state."

rfrandse, 2018-02-08:
<https://gerrit.openbmc-project.xyz/8914> Adding user friendly warning message in sensor page
fixes  #2748 OpenBMC GUI -> There should be appropriate and user friendly warning message has to display for "Sensors" Sub menu -> "FILTER BY SEVERITY" option

closed 2018-02-20 by geissonator