openbmc/openbmc #2749: How to modify the source code to fit my device

lockless, 2018-01-02:
My platform is as follows：
 AST2400
 use the following network interface:
  **RMII interface connect to Intel I350
  GMII interface connect PHY 88e1512**

First of all, I use palmetto platform(just because it's also ast2400).
The system run, but network can't work well.

```
    [   59.730000] 8021q: adding VLAN 0 to HW filter on device eth0
    [   59.730000] ftgmac100 1e660000.ethernet eth0: NCSI: Handler for packet type 0x82 returned -19
    [  OK  ] Started Network Service.
    [  OK  ] Started Set U-Boot environment variable.
    [  OK  ] Started Clear one time boot overrides.
    [  OK  ] Found device /dev/ttyS4.
    [  OK  ] Found device /dev/ttyVUART0.
    [   63.750000] ftgmac100 1e660000.ethernet eth0: NCSI: configuring channel 0
    [   63.750000] ftgmac100 1e660000.ethernet eth0: NCSI: Handler for packet type 0x8b returned -34
    [   63.850000] ftgmac100 1e660000.ethernet eth0: NCSI: Handler for packet type 0x8e returned -2
    [   63.850000] ftgmac100 1e660000.ethernet eth0: NCSI: channel 0 link down after config
    [   63.850000] ftgmac100 1e660000.ethernet eth0: NCSI interface dow
```

how and what source code i should modify to fit my device ?
can i get some advise?
thanks.

gwshan, 2018-01-02:
Please take a look on drivers/net/ethernet/farady/ftgmac100.c::ftgmac100_probe(). If the remote NIC doesn't support NCSI, you need remove the device-tree property "use-ncsi".

lockless, 2018-01-03:
Finally，Intel I350 with NCSI work well, so i will close this issue.

closed 2018-01-03 by lockless