openbmc/openbmc #2750: OpenBMC GUI - Not able to select the "warning" option under "Server health" Main menu -> "Sensors" sub menu

samahaba, 2018-01-02:
[image]

samahaba, 2018-01-02:
Steps to reproduce the problem
--------------------------------------

1. Go to "Server health" -> "Sensors"
2. Click on "All"
3. Click on "Critical"
4. Click on "Warning" and now observer that  there is no display shown which are belongs to "Warning" even though warning shown under "All" option

iftekharuli, 2018-02-08:
@samahaba this is how it is designed. If you are selecting "critical", it will show critical issues. If you select "warning", it will show warning issues.  

In the scenario above, you are selecting critical and then you are selecting warning. This means you are selecting "critical"**and** "warning". That is why "All" is being selected. Since All means critical and warning. 

I think what is being confusing here is- these items are not OR. These acts more like AND. 
In other word, think of these are checkboxes. Not radio buttons. 

[Karl](https://github.com/karlvochatzer) can probably explian more.

gtmills, 2018-03-15:
The user should be able to select "Critical" and "Warning" without being redirected to "All"

gtmills, 2018-03-15:
Go to "Server health" then try to select "Critical" and "Warning"

iftekharuli, 2018-03-15:
The way we designed this, we automatically select all when all the types are met. So, what do you suggest the third type would be? Can we categorize them as Normal? So that it is Critical, warning and Normal?

gtmills, 2018-03-15:
Yes, there would be 3 categories "Critical", "Warning", and "Normal". Not sure if "Normal" is the correct term and I don't know if the user would ever select "Normal".. Having "Normal" sensors displayed only when "All" is selected might be enough.

gtmills, 2018-03-15:
I am okay with adding a 3rd category for these "Normal" sensors. I am also okay with just allowing the user to select "Critical" and "Warning".

iftekharuli, 2018-03-19:
just allowing the user to select "Critical" and "Warning" breaks the design convention in the project, I will add normal button

rfrandse, 2018-03-20:
<https://gerrit.openbmc-project.xyz/9628> Adding new button 'Normal' in sensors page
Fixes  #2750 GUI: User should be able to select "Critical" and "Warning" Sensors (and not be redirected to "All")

iftekharuli, 2018-04-10:
Normal button has been added per discussion. This is how the GUI looks now. 

[image]

closed 2018-04-20 by geissonator