openbmc/openbmc #2752: IPMI DCMI : Get/Set (DHCP) configuration properties

dkodihal, 2018-01-03:
DHCP configuration properties - support only what's supported by systemd-networkd?

ratagupt, 2018-01-12:
- [ ] Activate DHCP Parameter 
        we can get the value of this parameter through reading of the dbus property (DHCPEnabled) for Dbus Object /xyz/openbmc_project/network/<interfacename>

- [ ] Discovery Configuration

       Bit 0 :- would be telling that DHCP Discover/Request packet would be having option 12.
       we need to enhance the dbus interface https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Network/DHCPConfiguration.interface.yaml and the networkd so that it can send the hostname.
     If this bit is enabled that means the network file should have SendHostname=true.
      By default it always true.

      Bit 1:- Vendor class information:- Enhance the  above dbus interface to include the vendor class identifier.
      If this bit is enabled that means the network file should have   VendorClassIdentifier="xxxx"
   
      Bit 7- Random Back off:- systemd-networkd doesn't support so we don't support it now.

- [ ] DHCP Timing 1,DHCP Timing 2,DHCP Timing 3:- Systemd-networkd doesn't allow to configure this,what we can do is to look at the systemd-networkd code and figure it out those values.So set for these parameters is NO-OP.

rfrandse, 2018-02-08:
<https://gerrit.openbmc-project.xyz/8883> DCMI: Add  get/set Configuration Parameters command
Resolves  #2752 IPMI DCMI : Get/Set (DHCP) configuration properties

closed 2018-02-22 by geissonator