openbmc/openbmc #2753: Why can‘t i open GUI？

lockless, 2018-01-03:
There is the position：
    the remote BMC ： AST2400， IP address 2.2.2.3
    local PC： Window 7 OS， IP address 2.2.2.2

now， bmc-----ping------>pc is ok！
            pc-------ping------>bmc  also ok！

but i can not open the OpenBMC GUI with the web browser：
 https://2.2.2.3:2443/login

can i get some advise？
thanks！

lockless, 2018-01-03:
should i download and setup some called "openbmc gui server" on my local pc?
if so,  where can i get this?

sivassrr, 2018-01-03:
https://github.com/openbmc/phosphor-webui - Download this and then follow the instruction from https://gerrit.openbmc-project.xyz/#/c/8360/1/README.md

Basically follow below steps once if you have done the above steps recommended in README.

1. https://<bmc ip>  in your browser
2. bypass the https security warning and it results in `Login required` JSON response
3. then use the same bmc IP address in the GUI web page to log in

Some time need to try for the 2nd time.

lockless, 2018-01-03:
thanks for your response, it works well. so i'll close this issue.

closed 2018-01-03 by lockless