openbmc/openbmc #2758: OpenBMC GUI - "Server power" status button and progress bar should sink up with ipl step of the system

samahaba, 2018-01-04:
[image]

samahaba, 2018-01-04:
Steps reproduce the problem
-----------------------------------
Make sure that system is in power off state
1. Go to "Server control" main menu -> Click on "Server power operation"
2. Click on "Power on" button and now observe that Status progress bar indicator suddenly move and status showing quickly as "Running" but at the back ground it is not the case still system is in ipling, so it will be good to match the progress bar as per the ipl step of the system and "Server status" button should show running when the system is completely power on that Running state at the back ground

samahaba, 2018-01-04:
The above expectation is repeat for all power option under "Server power operations"

gtmills, 2018-02-06:
The current status bar should match the state of /xyz/openbmc_project/state/host0

iftekharuli, 2018-02-08:
@samahaba @gtmills is there a rest call that can tell us the total percentage progress or how much is remaining? Is so, we can show and accurate progress bar. If not, we can show an infinite indeterminate state until we hear back from the server that it rebooted.

gtmills, 2018-02-08:
@iftekharuli There are 3 interfaces to look at for booting, "CurrentHostState", "BootProgress", and "OperatingSystemState".. The boot is complete when "OperatingSystemState" is "OperatingSystem.Status.OSStatus.BootComplete". When this happens "CurrentHostState" and "BootProgress" will be at "Running" and "OSStart" respectfully.  "CurrentHostState" to "Running" should happen slightly before "BootProgress" moves from "Unspecified".

A combination of all 3 will need to be used but "BootProgress" is probably the most useful for determining what state the boot is in. 
BootProgress is defined here: https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/State/Boot/Progress.interface.yaml
BootProgress  should go from "Unspecified" -> "MotherboardInit" -> "PCIInit" -> "OSStart"..
I have seen it loop though, so "Unspecified" -> "MotherboardInit" -> "PCIInit" -> "OSStart" then  "MotherboardInit" -> "PCIInit" -> "OSStart" a 2nd time. This loop might be caused by the host doing an auto reboot for reconfigure or SBE updates.. 

```
 curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET  https://${BMC_IP}/xyz/openbmc_project/state/host0
{
  "data": {
    "AttemptsLeft": 3, 
    "BootProgress": "xyz.openbmc_project.State.Boot.Progress.ProgressStages.OSStart", 
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Running", 
    "OperatingSystemState": "xyz.openbmc_project.State.OperatingSystem.Status.OSStatus.BootComplete", 
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.On"
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```

mdmillerii, 2018-02-08:
I think there is some overlap between hostboot and opal where both can issue overlapping boot progress codes, which may be an alternate cause of @gtmills observation.

Remembering from my prior p8 investigation, hostboot codes are driven from the IPL table, and the skiboot codes can be backtracked from the sensor update code.

rfrandse, 2018-04-18:
<https://gerrit.openbmc-project.xyz/10204> Add fixes for power operations
Resolves  #2758 OpenBMC GUI - "Server power" status button and progress bar should sink up with boot step of the system

closed 2018-04-26 by geissonator