openbmc/openbmc #2760: did openbmc have some command or tools for upgrade the firmware?

lockless, 2018-01-05:
I use the GUI to download ta new firmware like this：
[image]

then， there has flash-garrison image in dir /tmp/

$ ls /tmp/
**flash-garrison**                                                                     systemd-private-ea3e9f4c508543bbb3a7201b16a14026-systemd-timesyncd.service-zbrl2h
images

so， is there any command or tools i can use to upgrade the system to this version？

thanks！

mine260309, 2018-01-05:
The docs provides nice information about this:
https://github.com/openbmc/docs/tree/master/code-update

lockless, 2018-01-05:
thanks， it is helpful

closed 2018-01-05 by lockless