openbmc/openbmc #2762: Keep fan apps running after soft poweroff until shutdown

msbarth, 2018-01-05:
This is a continuation issue to enable what was done under #2629 where a system could be shutdown using a soft poweroff and the fan control apps must stay active until the system is completely off (in the case where the OS continues to offload jobs after the soft poweroff was issued).

- [ ] Update fan control service files to exit when poweroff is completed, not when started

msbarth, 2018-01-10:
Would like to have @bradbishop and @geissonator help in reviewing why the unit updates that I have made exit the fan services seem to always get started before the host and/or chassis has been powered off.

rfrandse, 2018-01-23:
<https://gerrit.openbmc-project.xyz/8577> Keep fan services running until poweroff completes
Resolves  #2762 Keep fan apps running after soft poweroff until shutdown

closed 2018-03-01 by geissonator