openbmc/openbmc #2763: Not getting power supply callout with master

geissonator, 2018-01-05:
I have a system (w54) that does not have one of it's power supplies plugged in.  I noticed we generate an error for it but doesn't have a callout.  This issue is to investigate this.

```
{
    "Resolved": 0, 
    "AdditionalData": [
        "_PID=1285"
    ], 
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error", 
    "Timestamp": 1515160025549, 
    "associations": [], 
    "Message": "org.open_power.Witherspoon.Fault.Error.PowerSupplyInputFault", 
    "Id": 5
}
```

bjwyman, 2018-01-05:
```
root@witherspoon:~# journalctl -b --no-pager --since "16:11:02" --until "16:11:03" -o verbose
-- Logs begin at Fri 2018-01-05 14:28:41 UTC, end at Fri 2018-01-05 16:36:10 UTC. --
Fri 2018-01-05 16:11:02.400071 UTC [s=f13e257418df485f9fda2b2b1b0d3842;i=4a758;b=593af607b13d41b9aa6adca528175769;m=1a2e2e69c;t=56209b1e5ac47;x=c78cfb006eb4dcb8]
    _BOOT_ID=593af607b13d41b9aa6adca528175769
    _MACHINE_ID=add14a8d18fe4f289ed865c90e27c7c2
    _HOSTNAME=witherspoon
    _KERNEL_SUBSYSTEM=platform
    _KERNEL_DEVICE=+platform:occ1-dev0
    _UDEV_SYSNAME=occ1-dev0
    _SOURCE_MONOTONIC_TIMESTAMP=7027530000
    _TRANSPORT=kernel
    PRIORITY=3
    SYSLOG_FACILITY=0
    SYSLOG_IDENTIFIER=kernel
    MESSAGE=occ-hwmon occ1-dev0: failed to get OCC poll response: -19
Fri 2018-01-05 16:11:02.392601 UTC [s=f13e257418df485f9fda2b2b1b0d3842;i=4a759;b=593af607b13d41b9aa6adca528175769;m=1a2e2f546;t=56209b1e5baf6;x=4c58c15330327642]
    _UID=0
    _GID=0
    _CAP_EFFECTIVE=3fffffffff
    _BOOT_ID=593af607b13d41b9aa6adca528175769
    _MACHINE_ID=add14a8d18fe4f289ed865c90e27c7c2
    _HOSTNAME=witherspoon
    _SYSTEMD_SLICE=system.slice
    PRIORITY=3
    SYSLOG_IDENTIFIER=kernel
    _TRANSPORT=syslog
    SYSLOG_FACILITY=1
    MESSAGE=[ 7027.530000] occ-hwmon occ1-dev0: failed to get OCC poll response: -19
    _PID=1030
    _COMM=klogd
    _EXE=/bin/busybox.nosuid
    _CMDLINE=/sbin/klogd -n
    _SYSTEMD_CGROUP=/system.slice/busybox-klogd.service
    _SYSTEMD_UNIT=busybox-klogd.service
    _SYSTEMD_INVOCATION_ID=e13f6836793b4cdcabd6c4c89c546dbe
    _SOURCE_REALTIME_TIMESTAMP=1515168662392601
Fri 2018-01-05 16:11:02.445748 UTC [s=f13e257418df485f9fda2b2b1b0d3842;i=4a75a;b=593af607b13d41b9aa6adca528175769;m=1a2e3a087;t=56209b1e6662f;x=f471eff97df0413f]
    _TRANSPORT=journal
    _UID=0
    _GID=0
    _CAP_EFFECTIVE=3fffffffff
    _BOOT_ID=593af607b13d41b9aa6adca528175769
    _MACHINE_ID=add14a8d18fe4f289ed865c90e27c7c2
    _HOSTNAME=witherspoon
    PRIORITY=3
    CODE_FILE=/usr/include/phosphor-logging/log.hpp
    CODE_LINE=100
    CODE_FUNC=helper_log
    MESSAGE=The power supply has indicated an input or under voltage condition. Check the power supply cabling and/or input power source.
    TRANSACTION_ID=2722497289
    RAW_STATUS=STATUS_WORD=0x2840|status0_input=0x8
    CALLOUT_INVENTORY_PATH=/xyz/openbmc_project/inventory/system/chassis/powersupply0
    SYSLOG_IDENTIFIER=witherspoon-psu-monitor
    _PID=1292
    _COMM=witherspoon-psu
    _EXE=/usr/sbin/witherspoon-psu-monitor
    _CMDLINE=witherspoon-psu-monitor --path=/sys/bus/i2c/devices/3-0069 --instance=0 --inventory=/system/chassis/powersupply0
    _SYSTEMD_CGROUP=/system.slice/system-witherspoon\x2dpower\x2dsupply\x2dmonitor.slice/witherspoon-power-supply-monitor@0.service
    _SYSTEMD_UNIT=witherspoon-power-supply-monitor@0.service
    _SYSTEMD_SLICE=system-witherspoon\x2dpower\x2dsupply\x2dmonitor.slice
    _SYSTEMD_INVOCATION_ID=02b561036e534993870d7d295ec18118
    _SOURCE_REALTIME_TIMESTAMP=1515168662445748
root@witherspoon:~#
```

spinler, 2018-01-05:
The problem is the empty AdditionalData entry in the log.

spinler, 2018-01-05:
@dkodihal Do you know why that may be missing?  I can see PowerSupplyInputFault with its metadata in both elog-lookup.cpp and elog-errors.hpp that phosphor-logging builds

spinler, 2018-01-09:
I can't recreate this when I build master.  @geissonator If you ever get the chance to check this again and it still doesn't work, I would like to see if this file exists, which is when we copy our error yaml so logging can find it:
```
sysroots/x86_64-linux/usr/share/phosphor-dbus-yaml/yaml/org/open_power/Witherspoon/Fault.errors.yaml
```

geissonator, 2018-01-09:
Tough to get the build images since they're done within docker containers and the output is lost.  I'm still seeing it on w54 though with latest code.
```
v2.0-162-gce17ef4
```

You can grab one of these images though from https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=witherspoon/lastSuccessfulBuild/artifact/deploy/images/witherspoon/

spinler, 2018-01-19:
I tried with master again, and it was fine:
```
  "data": {
    "AdditionalData": [
      "CALLOUT_INVENTORY_PATH=/xyz/openbmc_project/inventory/system/chassis/powersupply0", 
      "_PID=1223"
    ], 
    "Id": 669, 
    "Message": "xyz.openbmc_project.Inventory.Error.NotPresent", 
    "Resolved": 0, 
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error", 
    "Timestamp": 1516377252584, 
    "associations": [
      [
        "callout", 
        "fault", 
        "/xyz/openbmc_project/inventory/system/chassis/powersupply0"
      ]
    ]
  }
```

spinler, 2018-02-20:
I have yet to see problems with callouts at all with all of the testing both myself and @bjwyman have done, either using images built from master or driver releases.  The comment above shows that the callout was in the journal metadata, so about all that could happen is phosphor-logging couldn't get it for some reason.

closed 2018-02-20 by spinler

geissonator, 2018-02-20:
yeah, looks good now on w54:
```
[andrewg@gfwa123 ~]$ openbmc-events --server w54 view /xyz/openbmc_project/logging/entry/1
{
    "Resolved": 0, 
    "AdditionalData": [
        "CALLOUT_INVENTORY_PATH=/xyz/openbmc_project/inventory/system/chassis/powersupply0", 
        "RAW_STATUS=STATUS_WORD=0x2840|status0_input=0x8", 
        "_PID=1339"
    ], 
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error", 
    "Timestamp": 1519148632287, 
    "associations": [
        [
            "callout", 
            "fault", 
            "/xyz/openbmc_project/inventory/system/chassis/powersupply0"
        ]
    ], 
    "Message": "org.open_power.Witherspoon.Fault.Error.PowerSupplyInputFault", 
    "Id": 1
}

```