openbmc/openbmc #2764: Unable to load kernel intermittently on witherspoon

geissonator, 2018-01-05:
Seems specific to the UBI update process.  Something is happening to cause the boot variables to get out of synch, here's the trace from the udpate that led up to the fail:
```
Jan 05 23:02:10 witherspoon phosphor-version-software-manager[1230]: Untaring
Jan 05 23:02:10 witherspoon systemd[1]: Started SSH Per-Connection Server (9.41.165.103:48444).
Jan 05 23:02:11 witherspoon dropbear[2925]: Failed loading /var/lib/dropbear/dropbear_rsa_host_key
Jan 05 23:02:11 witherspoon dropbear[2925]: Failed loading /etc/dropbear/dropbear_dss_host_key
Jan 05 23:02:11 witherspoon dropbear[2925]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jan 05 23:02:11 witherspoon dropbear[2925]: Child connection from ::ffff:9.41.165.103:48444
Jan 05 23:02:11 witherspoon dropbear[2925]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:48444
Jan 05 23:02:12 witherspoon systemd[1]: Started SSH Per-Connection Server (9.41.165.103:48446).
Jan 05 23:02:12 witherspoon dropbear[2930]: Failed loading /var/lib/dropbear/dropbear_rsa_host_key
Jan 05 23:02:12 witherspoon dropbear[2930]: Failed loading /etc/dropbear/dropbear_dss_host_key
Jan 05 23:02:12 witherspoon dropbear[2930]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jan 05 23:02:12 witherspoon dropbear[2930]: Child connection from ::ffff:9.41.165.103:48446
Jan 05 23:02:13 witherspoon dropbear[2930]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:48446
Jan 05 23:02:14 witherspoon systemd[1]: Started SSH Per-Connection Server (9.41.165.103:48448).
Jan 05 23:02:14 witherspoon dropbear[2937]: Failed loading /var/lib/dropbear/dropbear_rsa_host_key
Jan 05 23:02:14 witherspoon dropbear[2937]: Failed loading /etc/dropbear/dropbear_dss_host_key
Jan 05 23:02:14 witherspoon dropbear[2937]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jan 05 23:02:14 witherspoon dropbear[2937]: Child connection from ::ffff:9.41.165.103:48448
Jan 05 23:02:15 witherspoon dropbear[2937]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:48448
Jan 05 23:02:16 witherspoon systemd[1]: Started SSH Per-Connection Server (9.41.165.103:48450).
Jan 05 23:02:16 witherspoon dropbear[2941]: Failed loading /var/lib/dropbear/dropbear_rsa_host_key
Jan 05 23:02:16 witherspoon dropbear[2941]: Failed loading /etc/dropbear/dropbear_dss_host_key
Jan 05 23:02:16 witherspoon dropbear[2941]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jan 05 23:02:16 witherspoon dropbear[2941]: Child connection from ::ffff:9.41.165.103:48450
Jan 05 23:02:16 witherspoon dropbear[2941]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:48450
Jan 05 23:02:19 witherspoon systemd[1]: Starting Set U-Boot environment variable...
Jan 05 23:02:19 witherspoon systemd[1]: Starting Set U-Boot environment variable...
Jan 05 23:02:19 witherspoon systemd[1]: Created slice system-obmc\x2dflash\x2dbmc\x2dupdateubootvars.slice.
Jan 05 23:02:20 witherspoon systemd[1]: Starting Updates the u-boot variable to point BMC version to 1be33b05...
Jan 05 23:02:20 witherspoon systemd[1]: Started Set U-Boot environment variable.
Jan 05 23:02:21 witherspoon systemd[1]: Started Set U-Boot environment variable.
Jan 05 23:02:24 witherspoon systemd[1]: Removed slice system-obmc\x2denable\x2dhost\x2dwatchdog.slice.
Jan 05 23:02:26 witherspoon kernel: watchdog: watchdog0: watchdog did not stop!
Jan 05 23:02:24 witherspoon systemd[1]: Removed slice system-mapper\x2dsubtree\x2dremove.slice.
Jan 05 23:02:24 witherspoon dropbear[2916]: Exit (root): Terminated by signal
Jan 05 23:02:24 witherspoon systemd[1]: Stopping SSH Per-Connection Server (9.41.165.103:48432)...
Jan 05 23:02:24 witherspoon dropbear[2930]: Exit (root): Terminated by signal
Jan 05 23:02:24 witherspoon systemd[1]: Stopping Phosphor sysfs LED controller...
Jan 05 23:02:24 witherspoon kernel[959]: [  813.340000] watchdog: watchdog0: watchdog did not stop!
Jan 05 23:02:24 witherspoon systemd[1]: Removed slice system-mboxd\x2dreload.slice.
Jan 05 23:02:24 witherspoon dropbear[2941]: Exit (root): Terminated by signal
Jan 05 23:02:24 witherspoon systemd[1]: Removed slice system-phosphor\x2dreboot\x2dhost.slice.
Jan 05 23:02:24 witherspoon dropbear[2937]: Exit (root): Terminated by signal
Jan 05 23:02:24 witherspoon systemd[1]: Stopping Phosphor sysfs LED controller...
Jan 05 23:02:25 witherspoon avahi-daemon[1000]: Got SIGTERM, quitting.
Jan 05 23:02:24 witherspoon systemd[1]: Stopped target Power0 Off.
Jan 05 23:02:25 witherspoon avahi-daemon[1000]: Leaving mDNS multicast group on interface eth0.IPv6 with address fe80::72e2:84ff:fe14:23e1.
Jan 05 23:02:24 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
Jan 05 23:02:25 witherspoon avahi-daemon[1000]: Leaving mDNS multicast group on interface eth0.IPv4 with address 9.3.86.161.
Jan 05 23:02:24 witherspoon systemd[1]: Removed slice system-phosphor\x2dreset\x2dhost\x2dcheck.slice.
Jan 05 23:02:25 witherspoon avahi-daemon[1000]: avahi-daemon 0.6.32 exiting.
Jan 05 23:02:24 witherspoon systemd[1]: Removed slice system-phosphor\x2dfan\x2dpresence\x2dtach.slice.
Jan 05 23:02:25 witherspoon kernel[959]: klogd: exiting
Jan 05 23:02:24 witherspoon systemd[1]: Stopped Assert bmc_booted LED.
Jan 05 23:02:28 witherspoon dropbear[2925]: Exit (root): Terminated by signal
Jan 05 23:02:24 witherspoon systemd[1]: Removed slice system-phosphor\x2dfan\x2dcontrol\x2dinit.slice.
Jan 05 23:02:24 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
Jan 05 23:02:24 witherspoon systemd[1]: Removed slice system-fsi\x2dbind.slice.
Jan 05 23:02:24 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
Jan 05 23:02:24 witherspoon systemd[1]: Removed slice system-phosphor\x2dfan\x2dcontrol.slice.
Jan 05 23:02:24 witherspoon systemd[1]: Removed slice system-op\x2dreset\x2dchassis\x2don.slice.
Jan 05 23:02:24 witherspoon systemd[1]: witherspoon-fan-watchdog.service: Killing process 1401 (watchdog) with signal SIGKILL.
Jan 05 23:02:24 witherspoon systemd[1]: Stopping Witherspoon Fan Watchdog Daemon...
Jan 05 23:02:24 witherspoon systemd[1]: Removed slice system-op\x2docc\x2denable.slice.
Jan 05 23:02:24 witherspoon systemd[1]: Stopping Phosphor sysfs LED controller...
Jan 05 23:02:24 witherspoon systemd[1]: Stopped Assert power_on LED.
Jan 05 23:02:24 witherspoon systemd[1]: Removed slice system-avsbus\x2denable.slice.
Jan 05 23:02:24 witherspoon systemd[1]: Removed slice system-avsbus\x2ddisable.slice.
[... 393 lines cut ...]
[    7.910000] [<805898c8>] (prepare_namespace) from [<805890b0>] (kernel_init_freeable+0x184/0x1cc)
[    7.910000] [<805890b0>] (kernel_init_freeable) from [<803fe480>] (kernel_init+0x18/0x104)
[    7.910000] [<803fe480>] (kernel_init) from [<8000a5e8>] (ret_from_fork+0x14/0x2c)

```

Here's the env variables in the fail state:
```
mtdparts=mtdparts=bmc:384k(u-boot),128k(u-boot-env),-(obmc-ubi)
netmask=255.255.254.0
obmc_bootcmd=ubi part obmc-ubi; run do_rwreset; ubi read ${loadaddr} ${kernelname}; bootm ${loadaddr}
partition=nor0,0
pnor-7a9ef57c=1
pnor-8764baa3=0
root=/dev/ubiblock4_0
serverip=9.3.164.219
set_bootargs=setenv bootargs console=ttyS4,115200n8 ubi.mtd=obmc-ubi,0,0,0 ubi.mtd=alt-obmc-ubi,0,0,4 ro rootfstype=squashfs ubi.block=${ubiblock} root=${root} 
spi_dma=yes
stderr=serial
stdin=serial
stdout=serial
ubiblock=0,1
verify=yes

```

mdmillerii, 2018-01-08:
Correction: You loaded the kernel just fine.

You just couldn't find the root file system from the kernel.

The env says root should be on alt, but the ubi block still says to make the block device on primary.

gkeishin, 2018-01-12:
We hit this again with the Master new build 1173
```
[   14.990000] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[   14.990000] CPU: 0 PID: 1 Comm: swapper Not tainted 4.10.17-72b52ebab574eaf0bcd2256ae16a25976a37c407 #1
[   14.990000] Hardware name: ASpeed SoC
[   14.990000] [<80010080>] (unwind_backtrace) from [<8000d934>] (show_stack+0x20/0x24)
[   14.990000] [<8000d934>] (show_stack) from [<801dbf2c>] (dump_stack+0x20/0x28)
[   14.990000] [<801dbf2c>] (dump_stack) from [<800a4a40>] (panic+0xc8/0x264)
[   14.990000] [<800a4a40>] (panic) from [<805894f8>] (mount_block_root+0x214/0x300)
[   14.990000] [<805894f8>] (mount_block_root) from [<80589740>] (mount_root+0x60/0x7c)
[   14.990000] [<80589740>] (mount_root) from [<805898c8>] (prepare_namespace+0x16c/0x1d0)
[   14.990000] [<805898c8>] (prepare_namespace) from [<805890b0>] (kernel_init_freeable+0x184/0x1cc)
[   14.990000] [<805890b0>] (kernel_init_freeable) from [<803fe480>] (kernel_init+0x18/0x104)
[   14.990000] [<803fe480>] (kernel_init) from [<8000a5e8>] (ret_from_fork+0x14/0x2c)
```

```
bootargs=console=ttyS4,115200n8 ubi.mtd=obmc-ubi,0,0,0 ubi.mtd=alt-obmc-ubi,0,0,4 ro rootfstype=squashfs
bootcmd=run set_bootargs; run obmc_bootcmd
bootdelay=2
c27712c9=2
do_rwreset=if test "${rwreset}" = "true"; then ubi remove rwfs; ubi create rwfs 0x400000; fi
e1478d95=2
e2ff887d=2
e7379c07=2
e8ea3001=2
ethact=aspeednic#0
ethaddr=70:e2:84:14:28:76
f6590ce0=2
fba43456=2
fefaeaa1=2
kernelname=kernel-aeedf969
loadaddr=80800000
mtdids=nor0=bmc
mtdparts=mtdparts=bmc:384k(u-boot),128k(u-boot-env),-(obmc-ubi)
obmc_bootcmd=ubi part obmc-ubi; run do_rwreset; ubi read ${loadaddr} ${kernelname}; bootm ${loadaddr}
pnor-498bd90d=0
pnor-df4b8673=0
root=/dev/ubiblock0_0
set_bootargs=setenv bootargs console=ttyS4,115200n8 ubi.mtd=obmc-ubi,0,0,0 ubi.mtd=alt-obmc-ubi,0,0,4 ro rootfstype=squashfs ubi.block=${ubiblock} root=${root}
spi_dma=yes
stderr=serial
stdin=serial
stdout=serial
ubiblock=4,0
verify=yes

Environment size: 1307/65531 bytes
ast#
```
Was able to recover by updating the kernel env

```
ast# setenv ubiblock 4,0
ast# setenv root /dev/ubiblock4_0
ast# saveenv
Saving Environment to Flash...
Un-Protected 1 sectors
Un-Protected 1 sectors
Erasing Flash...
. done
Erased 1 sectors
Writing to Flash... done
Protected 1 sectors
Protected 1 sectors
ast# 

and boot
```

geissonator, 2018-01-17:
Hit this again this evening, fixed with this:
```
setenv ubiblock 4,0
setenv root /dev/ubiblock4_0
saveenv
boot
```

Here's some data from the journal before the reboot:
```
Jan 17 01:15:03 witherspoon kernel: block ubiblock4_0: released
Jan 17 01:15:03 witherspoon kernel[996]: [ 1024.820000] block ubiblock4_0: released
Jan 17 01:16:09 witherspoon obmc-flash-bmc[3541]: Volume ID 0, size 257 LEBs (16809856 bytes, 16.0 MiB), LEB size 65408 bytes (63.9 KiB), static, name "rofs-8002b15e", alignment 1
Jan 17 01:17:10 witherspoon kernel: block ubiblock4_0: created from ubi4:0(rofs-8002b15e)
Jan 17 01:17:10 witherspoon kernel[996]: [ 1150.510000] block ubiblock4_0: created from ubi4:0(rofs-8002b15e)
Jan 17 01:17:21 witherspoon obmc-flash-bmc[3620]: Volume ID 0, size 65 LEBs (4251520 bytes, 4.1 MiB), LEB size 65408 bytes (63.9 KiB), static, name "kernel-8002b15e", alignment 1
Jan 17 01:17:28 witherspoon obmc-flash-bmc[3620]: Volume ID 3, size 65 LEBs (4251520 bytes, 4.1 MiB), LEB size 65408 bytes (63.9 KiB), static, name "kernel-8002b15e", alignment 1
Jan 17 01:17:38 witherspoon obmc-flash-bmc[3706]: [130B blob data]
Jan 17 01:17:39 witherspoon obmc-flash-bmc[3706]: [1.9K blob data]
Jan 17 01:17:39 witherspoon obmc-flash-bmc[3706]: [2.0K blob data]
Jan 17 01:17:39 witherspoon obmc-flash-bmc[3706]: [150B blob data]
Jan 17 01:17:39 witherspoon systemd[1]: Started Store read-only images 8002b15e to BMC storage.
Jan 17 01:17:39 witherspoon systemd[1]: Starting Set U-Boot environment variable...
Jan 17 01:17:39 witherspoon systemd[1]: Starting Set U-Boot environment variable...
Jan 17 01:17:39 witherspoon systemd[1]: Starting Set U-Boot environment variable...
Jan 17 01:17:40 witherspoon systemd[1]: Created slice system-obmc\x2dflash\x2dbmc\x2dupdateubootvars.slice.
Jan 17 01:17:40 witherspoon systemd[1]: Starting Updates the u-boot variable to point BMC version to 8002b15e...
Jan 17 01:17:40 witherspoon phosphor-image-updater[1159]: Error in Deleting image from image manager
Jan 17 01:17:43 witherspoon systemd[1]: Started Set U-Boot environment variable.
Jan 17 01:17:45 witherspoon systemd[1]: Started Set U-Boot environment variable.
Jan 17 01:17:45 witherspoon systemd[1]: Started Set U-Boot environment variable.
Jan 17 01:17:47 witherspoon dropbear[3533]: Exit (root): Terminated by signal
...
Jan 17 01:17:57 witherspoon systemd[1]: Stopped SSH Per-Connection Server (9.41.165.94:44998).
Jan 17 01:17:57 witherspoon systemd[1]: obmc-flash-bmc-updateubootvars@8002b15e.service: Main process exited, code=killed, status=15/TERM
Jan 17 01:17:57 witherspoon systemd[1]: Stopped Updates the u-boot variable to point BMC version to 8002b15e.
Jan 17 01:17:57 witherspoon systemd[1]: obmc-flash-bmc-updateubootvars@8002b15e.service: Unit entered failed state.
Jan 17 01:17:57 witherspoon systemd[1]: obmc-flash-bmc-updateubootvars@8002b15e.service: Failed with result 'signal'.
Jan 17 01:17:59 witherspoon systemd[1]: Removed slice system-obmc\x2dflash\x2dbmc\x2dupdateubootvars.slice.

```

geissonator, 2018-01-21:
CI system hit this again over the weekend, similar traces:
```
Jan 21 10:15:25 witherspoon systemd[1]: Starting Set U-Boot environment variable...
Jan 21 10:15:25 witherspoon systemd[1]: Starting Set U-Boot environment variable...
Jan 21 10:15:25 witherspoon systemd[1]: Created slice system-obmc\x2dflash\x2dbmc\x2dupdateubootvars.slice.
Jan 21 10:15:25 witherspoon systemd[1]: Starting Updates the u-boot variable to point BMC version to f9753315...
Jan 21 10:15:28 witherspoon systemd[1]: Started Set U-Boot environment variable.
Jan 21 10:15:28 witherspoon systemd[1]: Started Set U-Boot environment variable.
Jan 21 10:15:30 witherspoon systemd[1]: Removed slice system-obmc\x2dsync\x2dinventory\x2ditem.slice.
Jan 21 10:15:30 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
Jan 21 10:15:30 witherspoon systemd[1]: Stopping Phosphor Webserver...
Jan 21 10:15:30 witherspoon systemd[1]: Stopping Phosphor sysfs LED controller..
...
Jan 21 10:15:40 witherspoon systemd[1]: obmc-flash-bmc-updateubootvars@f9753315.service: Main process exited, code=killed, status=15/TERM
Jan 21 10:15:41 witherspoon systemd[1]: Stopped Updates the u-boot variable to point BMC version to f9753315.
Jan 21 10:15:41 witherspoon systemd[1]: obmc-flash-bmc-updateubootvars@f9753315.service: Unit entered failed state.
Jan 21 10:15:41 witherspoon systemd[1]: obmc-flash-bmc-updateubootvars@f9753315.service: Failed with result 'signal'.
Jan 21 10:15:42 witherspoon systemd[1]: Removed slice system-obmc\x2dflash\x2dbmc\x2dupdateubootvars.slice.
```

mdmillerii, 2018-01-22:
@geissonator please be sure to dump both primary and alternate u-boot environments when debugging this.

gkeishin, 2018-01-23:
How do we do that dumping the alt u but env @mdmillerii

anoo1, 2018-01-24:
It appears that code update is not waiting for the "obmc-flash-bmc-updateubootvars@.service" file, so the automation bucket can issue a reboot before it's done since the version would be marked as "Active".

anoo1, 2018-01-24:
Fix for "witherspoon phosphor-image-updater[1159]: Error in Deleting image from image manager":
https://gerrit.openbmc-project.xyz/#/c/8628/

rfrandse, 2018-01-25:
<https://gerrit.openbmc-project.xyz/8635> Wait for obmc-flash-bmc-updateubootvars@.service
Resolves  #2764 Unable to find root file system intermittently on System

closed 2018-01-26 by geissonator

anoo1, 2018-02-01:
> How do we do that dumping the alt u boot env @mdmillerii

```
fw_printenv -c /etc/alt_fw_env.config
```